pangsu0613/CLOCs #6: AP is very lower

alsun-oven, 2021-01-27:
I use  your model to eval. But the map is very low. only 72.12 in hard

pangsu0613, 2021-01-27:
This is the evaluation results that I got:
Predict_test:  False
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from ../model_dir/voxelnet-30950.tckpt
load existing model for fusion layer
Restoring parameters from /home/xi/for_submission/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
[100.0%][===================>][13.33it/s][04:39>00:00]     
generate label finished(13.49/s). start eval:
validation_loss: 0.14488257023894788
avg example to torch time: 1.179 ms
avg prep time: 8.701 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:99.33, 93.75, 92.89
bev  AP:96.51, 92.37, 89.41
3d   AP:92.74, 82.90, 77.75
aos  AP:99.17, 93.29, 92.11
Car AP@0.70, 0.50, 0.50:
bbox AP:99.33, 93.75, 92.89
bev  AP:99.38, 96.27, 95.56
3d   AP:99.55, 96.19, 95.34
aos  AP:99.17, 93.29, 92.11

Did you configure everything correctly based on the readme?

pangsu0613, 2021-01-27:
Predict_test:  False
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from ../model_dir/voxelnet-30950.tckpt
load existing model for fusion layer
Restoring parameters from /home/xi/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
[100.0%][===================>][12.63it/s][04:52>00:00]     
generate label finished(12.86/s). start eval:
validation_loss: 0.14488257023894788
avg example to torch time: 1.241 ms
avg prep time: 10.736 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:99.33, 93.75, 92.89
bev  AP:96.51, 92.37, 89.41
3d   AP:92.74, 82.90, 77.75
aos  AP:99.17, 93.29, 92.11
Car AP@0.70, 0.50, 0.50:
bbox AP:99.33, 93.75, 92.89
bev  AP:99.38, 96.27, 95.56
3d   AP:99.55, 96.19, 95.34
aos  AP:99.17, 93.29, 92.11

Did you configure everything correctly based on the readme?

pangsu0613, 2021-01-27:
This is the evaluation result that I got:
Predict_test: False
sparse_shape: [ 41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from ../model_dir/voxelnet-30950.tckpt
load existing model for fusion layer
Restoring parameters from /home/xi/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
[100.0%][===================>][12.63it/s][04:52>00:00]
generate label finished(12.86/s). start eval:
validation_loss: 0.14488257023894788
avg example to torch time: 1.241 ms
avg prep time: 10.736 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:99.33, 93.75, 92.89
bev AP:96.51, 92.37, 89.41
3d AP:92.74, 82.90, 77.75
aos AP:99.17, 93.29, 92.11
Car AP@0.70, 0.50, 0.50:
bbox AP:99.33, 93.75, 92.89
bev AP:99.38, 96.27, 95.56
3d AP:99.55, 96.19, 95.34
aos AP:99.17, 93.29, 92.11

Did you configure everything correctly based on the readme?

yinggo, 2021-02-08:
@pangsu0613 I train with the code, but the eval result is very low. Plz check my log.txt [log.txt](https://gist.github.com/yinggo/4482c87c16cd38a92f9c36454104448f), How can I train to get the high accuracy model? Should I change some parameters?

pangsu0613, 2021-02-08:
Hello @yinggo, I had a look at your log.txt, the numbers you got is quite weird, they are very close to 0.0. I guess maybe there is something wrong with the KITTI info pickle files or the evaluation script. Do you have original SECOND installed? If yes, you need to remove original SECOND directory from $PYTHONPATH and add CLOCs directory, or it may call some original SECOND functions inappropriately. Also, can you get good performance using my pretrained model?

yinggo, 2021-02-09:
@pangsu0613  thanks for your advice. I did not install second. I evaluate the pretrained model, it also get low accuracy:
```Generate output labels...
[100.0%][===================>][10.41it/s][05:56>00:00]     
generate label finished(10.57/s). start eval:
validation_loss: 325.96742813958514
avg example to torch time: 1.646 ms
avg prep time: 8.815 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:41.78, 36.01, 30.62
bev  AP:2.80, 2.73, 2.16
3d   AP:0.75, 0.69, 0.54
aos  AP:17.11, 15.19, 12.73
Car AP@0.70, 0.50, 0.50:
bbox AP:41.78, 36.01, 30.62
bev  AP:12.02, 11.42, 9.60
3d   AP:8.11, 7.81, 6.50
aos  AP:17.11, 15.19, 12.73

Car coco AP@0.50:0.05:0.95:
bbox AP:30.56, 26.72, 23.30
bev  AP:3.65, 3.54, 2.94
3d   AP:2.00, 1.93, 1.58
aos  AP:12.53, 11.22, 9.69
```
Maybe the evaluation script has some problem. But I did not changed the eval code before. That's kind of strange. I will debug and check the eval code.

pangsu0613, 2021-02-09:
Hi @yinggo , Ah, I see, I guess you didn't load the SECOND pretrained model so you are running an un-trained SECOND 3D detecor for fusion. We need to run a pretrained SECOND to generate 3D detection candidates for CLOCs. You need to download the weights from here: https://drive.google.com/drive/u/1/folders/1ScFUWPwzK5_VXb-LYQZuZVkiBj-dTMJ9, file "second_model.zip", I also put this link in the README. Unzip the file, and put the unzipped files (checkpoints.json and voxelnet-30950.tckpt) under directory /CLOCs/model_dir/, you could refer to Preperation, step 2 in the README for more detail.

yinggo, 2021-02-10:
@pangsu0613 That's well solved my problem. Thanks! Now I can get the same result as you present above.  I unzipped the files with the 'second_model' folder. Actually, the files should be directly put in the directory /CLOCs/model_dir/. A very stupid mistake. sorry. 
Further, I want to try PV-RCNN+Cascade R-CNN as the paper reported to achieve the 92.78%, 85.94% and 83.25% in easy, moderate and hard case.
I am not clear about  the README, Preperation, step 2, "Prepare the 3D detection candidates, run your 3D detector and save the results in the format that SECOND could read," in which variable should the results of 3D detector be saved and from where they should be read? I am not sure about the corresponding part in the code. Currently, in this code, I use the SECOND pretrained model to generate the 3D detection candidates, instead of directly use the 3D detection candidates. Could you give an instruction or an example for how to save and use the 3D detection candidate?

Joeless, 2022-02-26:
Hi~ @pangsu0613 
I evaluate it by your "CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt" , and get  a lower accuracy than you
can you tell why?

**My result:**
Predict_test:  False
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from ../model_dir/voxelnet-30950.tckpt
load existing model for fusion layer
Restoring parameters from /root/mym/git/CLOCs/model_dir/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
[100.0%][===================>][5.14it/s][16:53>00:00]      
generate label finished(3.71/s). start eval:
validation_loss: 0.15382743618570338
avg example to torch time: 3.456 ms
avg prep time: 72.179 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:99.33, 91.36, 88.33
bev  AP:96.16, 87.57, 84.28
3d   AP:89.89, 77.75, 72.10
aos  AP:99.18, 90.86, 87.39
Car AP@0.70, 0.50, 0.50:
bbox AP:99.33, 91.36, 88.33
bev  AP:99.43, 94.05, 91.20
3d   AP:99.39, 93.92, 91.00
aos  AP:99.18, 90.86, 87.39

Joeless, 2022-03-01:
Hi~  @pangsu0613 
Thanks for your answer 
I comment  line 466, and get lower AP than the original SECOND
My .config only changed the path  based on README.md
What's correct congfig?    My config is at the end of this comment~

**Now result:**
Predict_test:  False
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from ../model_dir/voxelnet-30950.tckpt
load existing model for fusion layer
Restoring parameters from /root/mym/git/CLOCs/model_dir/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
[100.0%][===================>][5.54it/s][14:01>00:00]      
generate label finished(4.47/s). start eval:
validation_loss: 0.15382743618570338
avg example to torch time: 1.951 ms
avg prep time: 54.293 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:96.09, 89.33, 84.80
bev  AP:94.78, 85.89, 81.46
3d   AP:88.27, 74.51, 69.43
aos  AP:95.96, 88.86, 84.02
Car AP@0.70, 0.50, 0.50:
bbox AP:96.09, 89.33, 84.80
bev  AP:98.04, 92.59, 89.76
3d   AP:97.96, 92.16, 87.61
aos  AP:95.96, 88.86, 84.02

**My car.fhd.config:**
model: {
  second: {
    voxel_generator {
      point_cloud_range : [0, -40, -3, 70.4, 40, 1]
      # point_cloud_range : [0, -32.0, -3, 52.8, 32.0, 1]
      voxel_size : [0.05, 0.05, 0.1]   # original is 0.05,0.05,0.1
      max_number_of_points_per_voxel : 5   # original is 5
    }

    voxel_feature_extractor: {
      module_class_name: "VoxelFeatureExtractorV3"
      num_filters: [16]
      with_distance: false
      num_input_features: 4
    }
    middle_feature_extractor: {
      module_class_name: "SpMiddleFHD"
      # num_filters_down1: [] # protobuf don't support empty list.
      # num_filters_down2: []
      downsample_factor: 8
      num_input_features: 4
    }
    rpn: {
      module_class_name: "RPNV2"
      layer_nums: [5, 5]
      layer_strides: [1, 2]
      num_filters: [128, 256]
      upsample_strides: [1, 2]
      num_upsample_filters: [256, 256]
      use_groupnorm: false
      num_groups: 32
      num_input_features: 128
    }
    loss: {
      classification_loss: {
        weighted_sigmoid_focal: {
          alpha: 0.25
[... 128 lines cut ...]
    use_moving_average: false
  }
  steps: 37120 #112215 #113715 #111360 # 619 * 50, super converge. increase this to achieve slightly better results original 30950
  steps_per_eval: 3712 #7481 # 619 * 5
  save_checkpoints_secs : 1800 # half hour 1800
  save_summary_steps : 10
  enable_mixed_precision: false # for fp16 training, don't use this.
  loss_scale_factor : 512.0
  clear_metrics_every_epoch: true
  detection_2d_path: "/root/*/git/CLOCs/d2_detection_data/data"
}

eval_input_reader: {
  batch_size: 1
  max_num_epochs : 160
  prefetch_size : 25
  max_number_of_voxels: 16000
  shuffle_points: false
  num_workers: 3
  anchor_area_threshold: -1
  remove_environment: false
  kitti_info_path: "/root/*/git/CLOCs/second/data/kitti_infos_val.pkl"
  #kitti_info_path: "/home/xi/KITTI_DATASET_ROOT/kitti_infos_test.pkl"
  kitti_root_path: "/root/*/git/CLOCs/second/data"
}

xumengqishuai, 2022-03-17:
hello,you are very great! could you please help me? my ap=0，I need your help!!!    @pangsu0613

Joeless, 2022-06-07:
> 

@pangsu0613 Hi  Sorry for ask you again.... Can you help me find the reason of lower AP?