pangsu0613/CLOCs #7: Help with inference for own dataset

xavidzo, 2021-01-28:
Hello, congrats on your nice work. I really enjoyed reading your paper and I am excited to try your code.
As I doing now my master thesis in 3d object detection, I would like to try CLOCs in combination with the state-of-the-art 3D lidar-based detector CenterPoint. Basically, in my project, I have access to camera images and lidar point clouds from sensors that are mounted on a bridge to detect vehicles on the highway, so the sensors are static. I would appreciate it if you could help me with some questions:

1) Do you think CLOCs is suitable for a real-time application? Only with CenterPoint theoretically, I could achieve 30 FPS

pangsu0613, 2021-01-28:
Thank you for your interests in CLOCs!
1. CLOCs fusion itself (the fusion layer) is a very small network, it is suitable for real-time application. CLOCs needs to run upon a 3D detector and a 2D detector, so for your application, you need to find a fast 3D detector (which you already have, CenterPoint) and a fast 2D detector, then running CLOCs on these two.
2. Yes, you need to have a pretrained 2D and 3D detectors trained by your custom dataset. Then you could use the pretrained 3D and 2D detector to generate the 2D and 3D detection candidates to train the CLOCs fusion layer. The final pipeline would be: (1) 2D and 3D detectors generate K 2D detection candidates and N 3D detection candidates (2) CLOCs takes the 2D and 3D detection candidates and do the fusion, the final output from CLOCs would be N fusesd confidence scores for N 3D detection candidates (3) For any 3D detector, it usually has a post processing stage (including NMS or other filtering) to filter the N detection candidates with confidence scores and generate the final predictions. We still need to do this, but with the CLOCs fused scores. I will add another figure in the README section to illustrate this pipeline.

xavidzo, 2021-02-03:
Thanks for your answer. I have more questions if you could help me please:

1. The code in the repository is almost entirely based on the SECOND implementation. The CLOCs functionality is contained mainly in the file master/second/pytorch/models/fusion.py, right? So if I want to use CenterPoint with CLOCs, when I build my model, I only have to add this fusion layer on top?
2. The 3D detector CenterPoint was trained on nuScenes, still have to retrain it on my own data. I don't know the number of detection candidates before NMS. Excuse my ignorance: Is this number of detection candidates in any 3D detector always a fixed number for every frame?
3. "run your 3D detector and save the results in the format that SECOND could read, including a matrix with shape of N by 7 that contains the N 3D bounding boxes, and a N-element vector for the 3D confidence scores". If I want to use CenterPoint, is this mandatory because the fusion layer only works with this kind of format?  Also, the N-element vector should only contain confidence scores, but what about the labels? I imagine a vector like [0.85, 0.66, 0.7] say there is 0.85 chance it is a car, 0.66 chance it's a truck, 0.7 is a pedestrian. But the vector values alone don't say anything about the labels, the labels don't play a role?
4. I guess using my custom dataset, I will have to project the 3D detected bounding boxes to the camera plane after I feed them to the fusion layer, or is this a step I should do before?
5. Regarding the /configs/car.fhd.config, the published code works for now only with the car class. Do I have to define a different config file for each class? In general what modifications should I make in the code for multi-class fusion?
6. 4. For me it's also not clear how the network might handle mutiple class / labels all at once. If the labels don't match the fusion process will not take place and the output will be null, so even though the 3D detector found some object with high confidence score? The output of CLOCs contains only confidence scores without class labels?
7. From my understanding, the 2D detections help to improve the 3D detections, but not the other way around. If the 2D detector finds an object with very high confidence, but the 3D detector does not, the fusion output will be null, correct?

pangsu0613, 2021-02-06:
I am glad to help.
1. fusion.py is just the fusion model, but you still need to prepare the input data for that model. The preparation functions are implemented in /second/pytorch/models/voxelnet.py and /second/utils/eval.py, and some other changes are made in train.py. If you build CLOCs on a 3D detector code based, such as SECOND or CenterPoint, you also need to disable the original data augmentations (in SECOND, they randomly add some ground truth vehicles from other frames during the training, rotate and change the size of each vehicles, etc). I believe CenterPoint do these type of data augmentation as well, you need to disable that, because these changes can not be done in the image plane. In a word, you can't directely add this fusion on the top of CenterPoint, there are some other changes needed.
2. I would say based on my knowledge, for most situations, it is fixed. It depends on the system structure of that 3D detector. If it is a one-stage detector, such as SECOND, the number would be quite large (tens of thousands), but it is fixed. For a two-stage or multi-stage detector, such as PointRCNN and PV-RCNN, for simplicity, CLOCs only take the detection candidates from the last stage, therefore, the number would be much smaller (hundreds), but it is still fixed. Even though for some 3D detectors it is not fixed, it is also fine since CLOCs use 1x1 convolution, but you need to modify the code a little bit.
3. The input for the fusion layer is just a tensor with 4 channels: [IoU between 2D and projected 3D bounding boxes, 2D confidence score, 3D confidence score, 3D distance between the object and the ego vehicle], the steps that I mentioned in README is the instructions on how to prepare that input tensor assuming you are using my codebase. For nuScenes dataset, they use very similar 7 parameters for each 3D bounding box as KITTI. The point is to build the input tensor for fusion layer, you could use your custom representations or formats in the intermedium steps to achieve that; The defination of label is based on the IoU between 3D detection candidates and ground truth boxes, IoU>=0.7 are positive targets, IoU<=0.5 are negative targets. I didn't use softmax, I have seperate models for different classes, so I only use binary crossentropy focal loss. For each detection candidate, it has one score in range [0,1], Actually, most 3D detectors do it in this way since it could achieve the best performance for each class. 
4. Partially answered in 3. Yes, you need to project the 3D bounding boxes to the image plane in order to build the input tensor for fusion layer.
5. For other classes such as pedestrian and cyclist, you could download the config files from SECOND-1.5 github repo, you only need to add the 'detection_2d_path:"../d2_detection_data"' in the same place as I did in car.fhd.config, and modify the KITTI dataset related directories, then you are good to go.
6. I haven't implemented a model that could predict multiple classes simultaneously since doing it seperately usually could achieve slightly better performance. For now, CLOCs works on each class seperately, one model for vehicles, one model for pedestrians and another model for cyclists. But extending from single-class to multi-class is not difficult, SECOND has already implement a multi-class model, basically, the output would be a vector instead of a number, then doing softmax to get the final result. 
7. Yes, because the final outputs are for 3D detections and we couldn't find a way to generate an accurate 3D detection from a 2D detection. The fusion does not consider objects only detected by 2D detector. Actually, based on our statistics on KITTI validation set, the number of objects detected only by 2D detector (missed by 3D detector) is very small compared to the total number of objects.

xavidzo, 2021-03-02:
Hello, thanks a lot again. Regarding your answer to the first point:

> fusion.py is just the fusion model, but you still need to prepare the input data for that model. The preparation functions are implemented in /second/pytorch/models/voxelnet.py and [...]

I don't get it why the data augmentation should be disabled for the 3D detector. I mean, this data augmentation is only done during training for better learning. Then, in evaluation mode for inference, this does not play a role anymore. So for the purpose of training the fusion layer of CLOCs, it only needs the results from the inference step, right? Or is it necessary to train the 3D detector in conjunction with the fusion layer at the same time? 
I thought one should train each 2D and 3D detector separately, then take the output of these detectors which serve as input for CLOCs and train the fusion layer apart and independently of the training of any detector?

pangsu0613, 2021-03-03:
Hello @xavidzo , your understanding is totally correct. The reason that I said I need to disable the data augmentation is that I used SECOND training config, pipeline and dataloader. While I was training CLOCs, I set the pretrained-SECOND net in eval (inference) mode, and I run pretrained-SECOND on the fly. So I also need the dataloader to provide point cloud for pretrained-SECOND to run inference, therefore, I have to disable the data augmentation in the original SECOND in order to feed pretrained-SECOND net with raw (unchanged) point cloud (data augmentation would change the raw point cloud). In a word, it is an implementation issue, because I put CLOCs training inside the original SECOND training pipeline. You could have your way to implement CLOCs training. I just said if you plan to do similar things like me, you need to disable the data augmentation. Let me know if you have further questions.

xavidzo, 2021-03-20:
Hello, hope you are doing great. So one question again about this step:

xavidzo, 2021-03-20:
Hello, hope you are doing great. So one question again about this step

> Step 3: Since the number of detection candidates are different for different 2D/3D detectors, you need to modify the corresponding parameters in the CLOCs code. Then train the CLOCs fusion. For example, there are 70400 (200x176x2) detection candidates in each frame from SECOND with batch size equals to 1. It is a very large number because SECOND is a one-stage detector, for other multi-stage detectors, you could just take the detection candidates before the final NMS function, that would reduce the number of detection candidates to hundreds or thousands.

a) Can you please explain where the numbers 200x176x2 in SECOND come from? I assume 70400 means the maximum number of detection candidates that can appear in one frame, but I guess more often even without NMS the detections are much less?
b) Also, in the dimensions k x n x 4  of the sparse input tensor T, this 70400 corresponds to the  n dimension, right? 
The k refers to the number of detections from the 2D detector, again how could I find out this number?
It still confuses me that this numbers _n_ and _k_ of detection candidates are fixed numbers... I thought these numbers could change with the input data (different images and different lidar frames may have more or less objects...?)
c) I am aware you are not the author of CenterPoint, I asked him about the number of 3D detection candidates before NMS, but he's too busy to answer. I assume this should be the portion of the code where this happens, can you check it please?
https://github.com/tianweiy/CenterPoint/blob/e26ea577c64825bfdab6598d1ff7a58438ee996b/det3d/models/bbox_heads/center_head.py#L472
Do you think I should set there the parameter _nms_iou_threshold_ as zero and then take as the maximum number of possible detection candidates the parameter  _nms_pre_max_size_ or _nms_post_max_size_.?? 
I see in the config file they have for nuscenes these values are  nms_iou_threshold = 0.2, nms_pre_max_size=1000 and nms_post_max_size=83

xavidzo, 2021-03-23:
Do you have any answer please?

pangsu0613, 2021-03-23:
Hello @xavidzo, sorry for the late response. 
(a) SECOND is an anchor-based one-stage approach. The whole field of view (lidar coordinate system, 0<x<70.4, -40<y<40) is divided into 176x200 grids, for each grid, there are 2 anchors. For each frame, there will be 70400 predictions (each anchor with a prediction), but most of them are negative examples with very low score. Here the post processing is very flexible, for original SECOND, they select the top 1000 predictions with the highest score for NMS (in config file, for #Postprocess, you can find some nms related thresholds). For my implementation, I just use all the 70400 detection candidates and let the fusion network to decide which to choose, because the fusion network can easily suppress these negative predictions. But, choosing the top 1000 is also good to CLOCs, I use 70400 for 2 reasons: one, I want to include as many detection candidates as possible to increase the recall; two, it is easier to implement. Again, this is very flexible, using other thresholds is perfectly fine with CLOCs, you don't have to use 70400, you can use top 1000, top 5000 and so on, even using detections after NMS is still ok, the CLOCs can still help you remove false positives.

(b) Whether k and n are fixed numbers depends on the types of the 2D/3D detectors. If a detector is a one-stage detector, like SECOND, the output for each frame will be fixed; While for two-stage or multi-stage detector, it is not fixed (Maybe for the first stage, it is, but for the rest stage, it is not). As I explained in (a), it is quite flexible here, you have the control of how many detection candidates to use. You can even choose different numbers for each frame, such as 100 for frame No.1 and 230 for frame No.2. As for implementation, you only need to set an upper bound, for example, for any frame, the number of detection candidates can not exceed 10000, then you can build the tensor with shape 10000, and set the rest as zeros. let's say you only have 5050 for a frame, just set the rest 4950 as zeros. For cascade-rcnn, I did similar things like this, I set an upper bound, so the fusion network can work.

(c) Yes, you are right, I think that is the spot. I am also working with CenterPoint, my plan is to try top 1000 or 5000 with no or very low score thresholding.

(d) Both the two scenarios you mentioned happened, and for both scenarios, the 3D detection score will be decreased, but it is hard to say how much it will be decreased. It is hard to remove these high-score 3D false positives as you mentioned, because based on our statistics, image-based detector tend to miss much more objects compared to LiDAR-based detector, so we can not punish these targets detected only by LiDAR too much, that would hurt the recall; Also, high-score false positives does not happen too much, for most of the reliable and good detectors, they don't have few high-score false positives, if they do, it means the nature design of the detector has some problems.

(e) We don't have any hard-code weights, I think this is the beatiful part of neural networks, it is data driven. There are so many differences for the ouptuts from different 2D/3D detectors, and there is no simple rule to decide the weight. That's why we choose using neural network to do the fusion, let the training process to decide everything.

pangsu0613, 2021-03-23:
Hello @xavidzo, sorry for the late response.
(a) SECOND is an anchor-based one-stage approach. The whole field of view (lidar coordinate system, 0<x<70.4, -40<y<40) is divided into 176x200 grids, for each grid, there are 2 anchors. For each frame, there will be 70400 predictions (each anchor with a prediction), but most of them are negative examples with very low score. Here the post processing is very flexible, for original SECOND, they select the top 1000 predictions with the highest score for NMS (in config file, for #Postprocess, you can find some nms related thresholds). For my implementation, I just use all the 70400 detection candidates and let the fusion network to decide which to choose, because the fusion network can easily suppress these negative predictions. But, choosing the top 1000 is also good to CLOCs, I use 70400 for 2 reasons: one, I want to include as many detection candidates as possible to increase the recall; two, it is easier to implement. Again, this is very flexible, using other thresholds is perfectly fine with CLOCs, you don't have to use 70400, you can use top 1000, top 5000 and so on, even using detections after NMS is still ok, the CLOCs can still help you remove false positives.

(b) Whether k and n are fixed numbers depends on the types of the 2D/3D detectors. If a detector is a one-stage detector, like SECOND, the output for each frame will be fixed; While for two-stage or multi-stage detector, it is not fixed (Maybe for the first stage, it is, but for the rest stage, it is not). As I explained in (a), it is quite flexible here, you have the control of how many detection candidates to use. You can even choose different numbers for each frame, such as 100 for frame No.1 and 230 for frame No.2. As for implementation, you only need to set an upper bound, for example, for any frame, the number of detection candidates can not exceed 10000, then you can build the tensor with shape 10000, and set the rest as zeros. let's say you only have 5050 for a frame, just set the rest 4950 as zeros. For cascade-rcnn, I did similar things like this, I set an upper bound, so the fusion network can work.

(c) Yes, you are right, I think that is the spot. I am also working with CenterPoint, my plan is to try top 1000 or 5000 with no or very low score thresholding.

(d) Both the two scenarios you mentioned happened, and for both scenarios, the 3D detection score will be decreased, but it is hard to say how much it will be decreased. It is hard to remove these high-score 3D false positives as you mentioned, because based on our statistics, image-based detector tend to miss much more objects compared to LiDAR-based detector, so we can not punish these targets detected only by LiDAR too much, that would hurt the recall; Also, high-score false positives does not happen too much, for most of the reliable and good detectors, they don't have too many high-score false positives, if they do, it means the nature design of the detector has some problems.

(e) We don't have any hard-code weights, I think this is the beatiful part of neural networks, it is data driven. There are so many differences for the ouptuts from different 2D/3D detectors, and there is no simple rule to decide the weight. That's why we choose using neural network to do the fusion, let the training process to decide everything.

xavidzo, 2021-03-24:
thank you very much @pangsu0613 . Greetings from Germany

pangsu0613, 2021-03-24:
You are welcome. Feel free to reopen this if you have further questions.

closed 2021-03-24 by pangsu0613

xavidzo, 2021-04-24:
Hello @pangsu0613,

xavidzo, 2021-04-26:
Can you please reopen this issue? @pangsu0613