pangsu0613/CLOCs #8: alternative dataset

dcy0577, 2021-01-28:
Hello, thank you for your wonderful work, I have some questions:
Q1 have you test CLOCs on the available other indoor dataset?
Q2 if we want to use own 3D 2D detector and CLOCs on custom dataset, can we also follow the pipeline that you mentioned in README,  or are there any other parts that needed to be modified?

pangsu0613, 2021-01-29:
Thank you for your interests in CLOCs!
1: No, I haven't tested CLOCs on any indoor dataset, I only tested on KITTI. But I think it should work, since from a fusion perspective, if you have some 3D detections in 3D space, and 2D detections in the image plane, and you know the calibration information to project the 3D detections to image plane, then you can perform CLOCs fusion.
2: Yes, you could follow my pipeline, but you need to write your own dataloader since SECOND is for KITTI. I built CLOCs based on SECOND because I started with SECOND and doing experiments on KITTI dataset (so I don't need to rebuild the wheels).

dcy0577, 2021-01-29:
Thank you for your quick reply. I'm not familiar with SECOND, but if I construct my own dataset in KITTI format, do I also need to write own dataloader?

pangsu0613, 2021-01-29:
No, if you transform your data into KITTI format, you don't need to write your own dataloader.

closed 2021-03-01 by pangsu0613