pangsu0613/CLOCs #10: CUDA error

yinggo, 2021-02-04:
I followed the instructions, but still meet some errors, please help.
I got RuntimeError: cuda runtime error (77)
What's the correct version of CUDA and CUDNN?

My environments:
```
$ conda list
# packages in environment at /home/ds1/anaconda3/envs/clocs:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
asn1crypto                0.22.0                   py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
blas                      1.0                         mkl    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
boost                     1.61.0                   py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
bzip2                     1.0.6                         3    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
ca-certificates           2021.1.19            h06a4308_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
certifi                   2016.2.28                py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
cffi                      1.10.0                   py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
cmake                     3.6.3                         0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
conda                     4.5.13                   py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
conda-env                 2.6.0                         0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
cryptography              1.8.1                    py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
cudatoolkit               9.0                  h13b8566_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
cudnn                     7.6.5                 cuda9.0_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
curl                      7.54.1                        0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
expat                     2.1.0                         0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
fire                      0.4.0                    pypi_0    pypi
freetype                  2.5.5                         2    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
icu                       54.1                          0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
idna                      2.6                      py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
imageio                   2.9.0                    pypi_0    pypi
intel-openmp              2020.2                      254    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
jbig                      2.1                           0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
jpeg                      9b                            0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
krb5                      1.13.2                        0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
libffi                    3.2.1                         1    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
libgcc-ng                 9.1.0                hdf63c60_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
libpng                    1.6.30                        1    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
libssh2                   1.8.0                         0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
libstdcxx-ng              9.1.0                hdf63c60_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
libtiff                   4.0.6                         3    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
llvmlite                  0.35.0                   pypi_0    pypi
mkl                       2020.2                      256    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
mkl-service               2.3.0            py36he8ac12f_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
mkl_fft                   1.2.0            py36h23d657b_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
mkl_random                1.1.1            py36h0573a6f_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
ncurses                   5.9                          10    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
networkx                  2.5                      pypi_0    pypi
ninja                     1.7.2                         0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
numba                     0.52.0                   pypi_0    pypi
numpy                     1.19.2           py36h54aff64_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
numpy-base                1.19.2           py36hfa32c7d_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
olefile                   0.44                     py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
openssl                   1.0.2l                        0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
packaging                 16.8                     py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
pillow                    4.2.1                    py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
pip                       9.0.1                    py36_1    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
protobuf                  3.14.0                   pypi_0    pypi
pybind11                  2.6.2                    pypi_0    pypi
pycosat                   0.6.3            py36h27cfd23_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
pycparser                 2.18                     py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
pyopenssl                 17.0.0                   py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
pyparsing                 2.2.0                    py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
python                    3.6.2                         0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
pytorch                   1.1.0           cuda90py36h8b0c50b_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/main
pywavelets                1.1.1                    pypi_0    pypi
readline                  6.2                           2    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
requests                  2.14.2                   py36_0    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
ruamel_yaml               0.11.14                  py36_1    https://mirrors.tuna.tsinghua.edu.cn/anaconda/pkgs/free
scikit-image              0.17.2                   pypi_0    pypi
[... 282 lines cut ...]
Traceback (most recent call last):
  File "./second/pytorch/train.py", line 920, in <module>
    fire.Fire()
  File "/home/ds1/anaconda3/envs/clocs/lib/python3.6/site-packages/fire/core.py", line 141, in Fire
    component_trace = _Fire(component, args, parsed_flag_args, context, name)
  File "/home/ds1/anaconda3/envs/clocs/lib/python3.6/site-packages/fire/core.py", line 471, in _Fire
    target=component.__name__)
  File "/home/ds1/anaconda3/envs/clocs/lib/python3.6/site-packages/fire/core.py", line 681, in _CallAndUpdateTrace
    component = fn(*varargs, **kwargs)
  File "./second/pytorch/train.py", line 107, in train
    torch.manual_seed(3)
  File "/home/ds1/anaconda3/envs/clocs/CLOCs/torchplus/train/checkpoint.py", line 173, in save_models
    save(model_dir, model, name, global_step, max_to_keep, keep_latest)
  File "/home/ds1/anaconda3/envs/clocs/CLOCs/torchplus/train/checkpoint.py", line 90, in save
    torch.save(model.state_dict(), ckpt_path)
  File "/home/ds1/anaconda3/envs/clocs/lib/python3.6/site-packages/torch/serialization.py", line 224, in save
    return _with_file_like(f, "wb", lambda f: _save(obj, f, pickle_module, pickle_protocol))
  File "/home/ds1/anaconda3/envs/clocs/lib/python3.6/site-packages/torch/serialization.py", line 149, in _with_file_like
    return body(f)
  File "/home/ds1/anaconda3/envs/clocs/lib/python3.6/site-packages/torch/serialization.py", line 224, in <lambda>
    return _with_file_like(f, "wb", lambda f: _save(obj, f, pickle_module, pickle_protocol))
  File "/home/ds1/anaconda3/envs/clocs/lib/python3.6/site-packages/torch/serialization.py", line 303, in _save
    serialized_storages[key]._write_file(f, _should_read_directly(f))
RuntimeError: cuda runtime error (77) : an illegal memory access was encountered at ../torch/csrc/generic/serialization.cpp:23
```

pangsu0613, 2021-02-04:
I have tested on CUDA 9,10 and even 11, but I didn't see this error before. Looks like the spconv is not installed properly, can you run SECOND successfully?

BTW, you could add the following things at the beginning of the code to avoid these NUMBA warnings
from numba.core.errors import NumbaDeprecationWarning, NumbaPendingDeprecationWarning,NumbaPerformanceWarning,NumbaWarning
import warnings
warnings.simplefilter('ignore', category=NumbaDeprecationWarning)
warnings.simplefilter('ignore', category=NumbaPendingDeprecationWarning)
warnings.simplefilter('ignore', category=NumbaPerformanceWarning)
warnings.simplefilter('ignore', category=NumbaWarning)
warnings.simplefilter('ignore')
warnings.filterwarnings('ignore')

yinggo, 2021-02-08:
Thanks for your kind reply. It solved.

closed 2021-02-08 by yinggo