pangsu0613/CLOCs #12: Can't open file

harris-mohamed, 2021-02-11:


pangsu0613, 2021-02-11:
Hello @harris-mohamed, you need to run the training or evaluation command in your terminal under /CLOCs/second directory, because train.py is under /CLOCs/second/pytorch.

harris-mohamed, 2021-02-11:
I was able to run it a little further, but now I am getting 
`line 307, in _read_and_prep_v9
    v_path = pathlib.Path(root_path) / info['velodyne_path']
KeyError: 'velodyne_path'`

As far as I can tell, this is an issue with my config file but I have manually checked that they are pointed to the KITTI dataset and that my dataset is set up in the way specified.

pangsu0613, 2021-02-11:
hello @harris-mohamed, if your config file is good, I think this is due to the setup for the KITTI dataset directory or the kitti pickle info files. You need to setup the /KITTI_DATASET_ROOT directory exactely the same as I specified in the README ("Prepare Dataset (KITTI)" section). You need to download the velodyne point clouds and other related data from KITTI website, and download the pickle files from my google drive shared folder, then put everything in the right directory.

harris-mohamed, 2021-02-12:
Thanks for the response. I am still having the same issue, and I am baffled as to what my issue could be. Attached is a screenshot of my directory (the velodyne_reduced I downloaded from your google drive) as well as my config file. To me, it seems that it matches what you have in the readme. 
[image]
[car.fhd.config.txt](https://github.com/pangsu0613/CLOCs/files/5969188/car.fhd.config.txt)

pangsu0613, 2021-02-12:
Based on your error information, this means that the code cannot find the kitti info pickle file, I had a look at your config file, you are using relative directory that points to the wrong directory, I recommend using full directory for all the info and dataset root path, such as /home/yourname/CLOCs/second/KITTI_DATASET_ROOT/ and /home/yourname/CLOCs/second/KITTI_DATASET_ROOT/kitti_infos_train.pkl for your situation.

harris-mohamed, 2021-02-12:
I switched all of the paths to full paths, but I am still having the same issue. One difference that I noted is that I pulled the repo onto an external drive due to the dataset size, so that all of my paths are "/media/......" instead of "/home/....". Is this what could be causing this issue?

harris-mohamed, 2021-02-17:
Hello, 

Are there any other ideas with this? I tried pulling the repo in a fresh location, but I still have the same issues. One difference I did note is that my velodyne test has 7518 samples instead of 7580 samples. Could this be the issue?

pangsu0613, 2021-02-17:
In KITTI, there should be 7481 samples for training, 7518 samples for testing. Do you still have the KeyError: 'velodyne_path' error? If so, that means the code didn't load the pickle info files properly. Maybe the pickle info files you downloaded are damaged, the size of "kitti_infos_train.pkl" is 12598018 bytes, "kitti_dbinfos_train.pkl" is 6983252 bytes, "kitti_infos_val.pkl" is 13016563 bytes. Are these same as yours, if not, I suggest redownload them. Also, you could write a small script to load and read these pickle files and see what's going on.
with open(info_path, 'rb') as f:
      infos = pickle.load(f)

harris-mohamed, 2021-02-17:
Upon checking the pkl files, it seems that there was something wrong with my downloaded files. Redownloading it fixed this issue, but now I am getting 
```
Traceback (most recent call last):
  File "./pytorch/train.py", line 22, in <module>
    from second.utils.eval import get_coco_eval_result, get_official_eval_result,bev_box_overlap,d3_box_overlap
ImportError: cannot import name 'd3_box_overlap'
```
However, my bashrc is adding the python paths that were specified in the setup. Is there something else it needs?

pangsu0613, 2021-02-17:
@harris-mohamed  'd3_box_overlap' is one of the functions in /CLOCs/second/utils/eval.py, did you modify the file? or after you modify the ~/.bashrc, did you source it?

harris-mohamed, 2021-02-17:
I did not modify it, and I sourced the bashrc. I see the function clearly defined in the eval.py file, but I am at a loss as to why I cannot import it. I can import all of the other functions in that line except d3_box_overlap

pangsu0613, 2021-02-17:
That's weird, if you can import other functions, then there is no reason that you cannot import d3_box_overlap, could you show me your $PYTHONPATH, train.py and eval.py?

harris-mohamed, 2021-02-18:
I also included my directory setup in case that helps.
[image]

[image]

[train_eval.zip](https://github.com/pangsu0613/CLOCs/files/5999316/train_eval.zip)

pangsu0613, 2021-02-18:
@harris-mohamed   thank you. I think there is no problem with your train.py and eval.py, they are the same as mine. That means the problem highly likely locates in the python environment, could you remove the "..../CLOCS_TEST/CLOCs/second" and ".../CLOCS_TEST/second.pytorch" from your $PYTHONPATH and see if that helps (only keep /CLOCs_TEST/CLOCs), there are many '/second' directories, and not all contains utils/eval.py, so I guess that could be the issue.

harris-mohamed, 2021-02-18:
That fixed my error, but now I am having an issue with the script itself. It might be a long shot, but have you seen the error below before? Should I set the block_filtering to true?
```

Traceback (most recent call last):
  File "./pytorch/train.py", line 918, in <module>
    fire.Fire()
  File "/media/harris/Projects/anaconda/envs/clocs_test/lib/python3.6/site-packages/fire/core.py", line 141, in Fire
    component_trace = _Fire(component, args, parsed_flag_args, context, name)
  File "/media/harris/Projects/anaconda/envs/clocs_test/lib/python3.6/site-packages/fire/core.py", line 471, in _Fire
    target=component.__name__)
  File "/media/harris/Projects/anaconda/envs/clocs_test/lib/python3.6/site-packages/fire/core.py", line 681, in _CallAndUpdateTrace
    component = fn(*varargs, **kwargs)
  File "./pytorch/train.py", line 656, in evaluate
    for example in iter(eval_dataloader):
  File "/media/harris/Projects/anaconda/envs/clocs_test/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 435, in __next__
    data = self._next_data()
  File "/media/harris/Projects/anaconda/envs/clocs_test/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 475, in _next_data
    data = self._dataset_fetcher.fetch(index)  # may raise StopIteration
  File "/media/harris/Projects/anaconda/envs/clocs_test/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/media/harris/Projects/anaconda/envs/clocs_test/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/media/harris/Projects/CLOCS_TEST/CLOCs/second/pytorch/builder/input_reader_builder.py", line 18, in __getitem__
    return self._dataset[idx]
  File "/media/harris/Projects/CLOCS_TEST/CLOCs/second/data/dataset.py", line 70, in __getitem__
    prep_func=self._prep_func)
  File "/media/harris/Projects/CLOCS_TEST/CLOCs/second/data/preprocess.py", line 363, in _read_and_prep_v9
    example = prep_func(input_dict=input_dict)
  File "/media/harris/Projects/CLOCS_TEST/CLOCs/second/data/preprocess.py", line 225, in prep_pointcloud
    points, max_voxels)
  File "/media/harris/Projects/anaconda/envs/clocs_test/lib/python3.6/site-packages/spconv/utils/__init__.py", line 173, in generate
    or self._max_voxels, self._full_mean)
  File "/media/harris/Projects/anaconda/envs/clocs_test/lib/python3.6/site-packages/spconv/utils/__init__.py", line 69, in points_to_voxel
    assert block_filtering is False
AssertionError

```

pangsu0613, 2021-02-19:
@harris-mohamed , This is because the version of spconv is not correct, you need to install the commit that I put in the readme (spconv-v1.0, commit 8da6f96) .

pangsu0613, 2021-02-20:
#15  this guy had the same issue and has solved by reinstall spconv-v1.0 as I suggested. Hopes it helps.

shaunkheng97, 2021-02-22:
This may help,
https://github.com/open-mmlab/OpenPCDet/issues/178#issuecomment-663194520

after `git checkout 8da6f967fb9a054d8870c3515b1b44eca2103634`, continue with 
3. Ensure you have install pytorch 1.0 in your environment, run `python setup.py bdist_wheel` (don't use python setup.py install).
4. Run `cd ./dist`, use pip to install generated whl file.

If you encounter some setup error which I did, this may help:
https://github.com/traveller59/spconv/issues/104#issuecomment-583262965

harris-mohamed, 2021-03-01:
Thank you everyone for your detailed help. I was successfully able to run the pretrained model.

closed 2021-03-01 by harris-mohamed