pangsu0613/CLOCs #13: spconv.units cannot import "non_max_suppression"

shaunkheng97, 2021-02-11:
[image]

Does anyone have this issue after installing spconv?

pangsu0613, 2021-02-16:
I remembered this mistake when I set a conda environment long time ago, it could be the pytorch and cuda is not installed properly. first, you need to check the return of 'torch.cuda.is_available()', if it returns False, it means that the your pytorch and cuda has some issues, probably they are imcompatible, you need to reinstall pytorch; if it returns True, then it could be other problems.

closed 2021-02-18 by shaunkheng97

Anirbanbhk88, 2022-04-01:
@shaunkheng97 I am also facing this issue. How did you solve it?