pangsu0613/CLOCs #15: Assertion Error

shaunkheng97, 2021-02-18:
I am facing this Assertion Error, not sure if this is a problem for my packages?

[image]
[image]

shaunkheng97, 2021-02-18:
I have the similar issue when I am training and when I am running the inference on the pretrained cascade_rcnn x seconds

pangsu0613, 2021-02-19:
This is because the version of spconv is not correct, you need to use the commit that I put in the readme (spconv-v1.0, commit 8da6f96) .

shaunkheng97, 2021-02-20:
Install v1.0 and I solved this issue, thanks!

closed 2021-02-20 by shaunkheng97

wj-shin, 2021-03-25:
I look ths page so I install spconv version 1.0.0 by modify setup.py version 1.2.1 to 1.0.0 but I have same error

wj-shin, 2021-03-25:
shaunkheng97
ho do you install v1.0 I struggled to install v1.0 but I can not install v1.0 

I want your tip.

shaunkheng97, 2021-03-25:
https://github.com/pangsu0613/CLOCs/issues/12#issuecomment-783059639

You can refer to this link

wj-shin, 2021-03-25:
I will try this right now thanks your comments🥰

wj-shin, 2021-03-25:
I do https://github.com/open-mmlab/OpenPCDet/issues/178#issuecomment-663194520 this link already..

But this link for me.. thaks a lot

wj-shin, 2021-03-26:
Hey Shaunkheng
I finished installing and try again doesn't work...
Can I know why It is

shaunkheng97, 2021-03-26:
What problem did you face?

wj-shin, 2021-03-26:
Traceback (most recent call last):
  File "./pytorch/train.py", line 918, in <module>
    fire.Fire()
  File "/home/******/anaconda3/envs/CLOCs/lib/python3.6/site-packages/fire/core.py", line 141, in Fire
    component_trace = _Fire(component, args, parsed_flag_args, context, name)
  File "/home/******/anaconda3/envs/CLOCs/lib/python3.6/site-packages/fire/core.py", line 471, in _Fire
    target=component.__name__)
  File "/home/******/anaconda3/envs/CLOCs/lib/python3.6/site-packages/fire/core.py", line 681, in _CallAndUpdateTrace
    component = fn(*varargs, **kwargs)
  File "./pytorch/train.py", line 371, in train
    raise e
  File "./pytorch/train.py", line 231, in train
    example = next(data_iter)
  File "/home/******/anaconda3/envs/CLOCs/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 518, in __next__
    data = self._next_data()
  File "/home/******/anaconda3/envs/CLOCs/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 1200, in _next_data
    return self._process_data(data)
  File "/home/******/anaconda3/envs/CLOCs/lib/python3.6/site-packages/torch/utils/data/dataloader.py", line 1226, in _process_data
    data.reraise()
  File "/home/******/anaconda3/envs/CLOCs/lib/python3.6/site-packages/torch/_utils.py", line 434, in reraise
    raise self.exc_type(msg)
AssertionError: Caught AssertionError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/******/anaconda3/envs/CLOCs/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 202, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/*****/anaconda3/envs/CLOCs/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/*****/anaconda3/envs/CLOCs/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/******/wj-shin/FOR_CLOCs/CLOCs/second/pytorch/builder/input_reader_builder.py", line 18, in __getitem__
    return self._dataset[idx]
  File "/home/*****/wj-shin/FOR_CLOCs/CLOCs/second/data/dataset.py", line 70, in __getitem__
    prep_func=self._prep_func)
  File "/home/*****/wj-shin/FOR_CLOCs/CLOCs/second/data/preprocess.py", line 363, in _read_and_prep_v9
    example = prep_func(input_dict=input_dict)
  File "/home/*****/wj-shin/FOR_CLOCs/CLOCs/second/data/preprocess.py", line 225, in prep_pointcloud
    points, max_voxels)
  File "/home/*****/anaconda3/envs/CLOCs/lib/python3.6/site-packages/spconv/utils/__init__.py", line 166, in generate
    or self._max_voxels, self._full_mean)
  File "/home/****/anaconda3/envs/CLOCs/lib/python3.6/site-packages/spconv/utils/__init__.py", line 63, in points_to_voxel
    assert block_filtering is False
AssertionError

***** is related my private things so chnge to *

wj-shin, 2021-03-26:
Is it problem about kitti datasets?

wj-shin, 2021-03-26:
I do train second v1.16 already

So  can  I  do clocs on second v1.16 insted of second v1.15?

pangsu0613, 2021-03-26:
Hello @wj-shin , I guess you mean you have trained SECOND-V1.6, there is only 4 versions of SECOND, V1.0,V1.5,V1.51 and V1.6. V1.0, V1.5 and V1.6 are quite different in many detail parts. Currently CLOCs is based on SECOND-V1.5, so V1.6 is not gonna work. Also, you don't need to train SECOND-V1.5, I have provide the pretrained SECOND-V1.5 weights, you could just use the links in readme to download it.

wj-shin, 2021-03-26:
> Hello @wj-shin , I guess you mean you have trained SECOND-V1.6, there is only 4 versions of SECOND, V1.0,V1.5,V1.51 and V1.6. V1.0, V1.5 and V1.6 are quite different in many detail parts. Currently [...]

Right that was my mistake.
Hmm.... do you mean I evaluate pretrained model which is provided by you?
I do this right now

wj-shin, 2021-03-26:
I have a same error if I evaluate pretrained model...

pangsu0613, 2021-03-26:
ok, I see. I thought you are trying SECOND-V1.6 model. Anyway, what is your spconv version? you need to install the specific commit (spconv-v1.0, commit 8da6f96), have you succeed in this?

wj-shin, 2021-03-26:
# packages in environment at /envs/CLOCs:
#
# Name                    Version                   Build  Channel
spconv                    1.0                      pypi_0    pypi

I install spcov 1.0 by this link https://github.com/traveller59/spconv/issues/104#issuecomment-583262965

wj-shin, 2021-03-26:
Is this a problem if i install  by that link?

pangsu0613, 2021-03-26:
So, I guess you installed spconv commit 5df9738, not commit 8da6f96. I think you need to switch to commit 8da6f96

wj-shin, 2021-03-26:
> So, I guess you installed spconv commit 5df9738, not commit 8da6f96. I think you need to switch to commit 8da6f96

I do right now!!
Sorry

pangsu0613, 2021-03-26:
no problem, I understand configing spconv is kind of annoying, we plan to remove it in the future.

wj-shin, 2021-03-26:
You mean that , in the future, Clocs does not occur this error regardless of spconv version?

pangsu0613, 2021-03-26:
yes, I think I will have some one in the lab to help me to improve the code in the near future. we plan to remove unrelated dependencies. Now we need spconv because we need to run SECOND on the fly.

wj-shin, 2021-03-26:
@pangsu0613 thanks a lot I hope you have a nice day

wj-shin, 2021-03-26:
And I success train really thanks a lot @pangsu0613