pangsu0613/CLOCs #16: Error on AP score after training/evaluation on SecCas

shaunkheng97, 2021-02-20:
[image]

Had this problem after I train, not sure why does the AP are 0

pangsu0613, 2021-02-20:
The error you show here is related to the calculation of COCO AP, which is not necessary since we only care about KITTI evaluation metric, you could simpley delete or comment the lines ''result = get_coco_eval_result(gt_annos, dt_annos, class_names)'' and related variables ('result') in trian.py (I remembered there are three places). But this error is not related to 0.0 AP issue, I guess the 0.0 AP issue is related to the evaluation script and/or the pickle info files or others. The size of "kitti_infos_train.pkl" is 12598018 bytes, "kitti_dbinfos_train.pkl" is 6983252 bytes, "kitti_infos_val.pkl" is 13016563 bytes, you could compare yours with these and see if they have been downloaded correctly. Did you modify the other places of the code? If you could sent me your train.py, config file, and /CLOCs/second/utils/eval.py,  I could have a look and see what I can do.

shaunkheng97, 2021-02-21:
All 3 .pkl files were correctly downloaded. I only modified the codes for the directories. Do I have to write another function for kitti_eval?

shaunkheng97, 2021-02-22:
Hi Su Pang, this is still the results I've gotten from running SecCas Pretrained:
[image]

These are the modifications that I have made for train.py as mentioned in the email:
[image]

[image]

But still, these are the results that I got:
[image]

pangsu0613, 2021-02-22:
Could you show me all the output from your terminal, it looks like something happens in eval.py.

shaunkheng97, 2021-02-22:
```
Predict_test:  False
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from /hpctmp/e0318980/CLOCs/model_dir/voxelnet-30950.tckpt
load existing model for fusion layer
Restoring parameters from /hpctmp/e0318980/CLOCs/model_dir/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
/hpctmp/e0318980/CLOCs/second/core/geometry.py:146: NumbaWarning:
Compilation is falling back to object mode WITH looplifting enabled because Function "points_in_convex_polygon_jit" failed type infe              rence due to: No implementation of function Function(<built-in function getitem>) found for signature:

 >>> getitem(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))

There are 22 candidate implementations:
   - Of which 20 did not match due to:
   Overload of function 'getitem': File: <numerous>: Line N/A.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    No match.
   - Of which 2 did not match due to:
   Overload in function 'GetItemBuffer.generic': File: numba/core/typing/arraydecl.py: Line 162.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    Rejected as the implementation raised a specific error:
      TypeError: unsupported array index type list(int64)<iv=None> in Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>)
  raised from /home/svu/e0318980/.local/lib/python3.6/site-packages/numba/core/typing/arraydecl.py:69

During: typing of intrinsic-call at /hpctmp/e0318980/CLOCs/second/core/geometry.py (162)

File "../core/geometry.py", line 162:
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
        vec1 = polygon - polygon[:, [num_points_of_polygon - 1] +
                                 list(range(num_points_of_polygon - 1)), :]
                                 ^

  @numba.jit
/hpctmp/e0318980/CLOCs/second/core/geometry.py:146: NumbaWarning:
Compilation is falling back to object mode WITHOUT looplifting enabled because Function "points_in_convex_polygon_jit" failed type i              nference due to: Cannot determine Numba type of <class 'numba.core.dispatcher.LiftedLoop'>

File "../core/geometry.py", line 170:
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
    cross = 0.0
    for i in range(num_points):
    ^

  @numba.jit
/home/svu/e0318980/.local/lib/python3.6/site-packages/numba/core/object_mode_passes.py:152: NumbaWarning: Function "points_in_convex              _polygon_jit" was compiled in object mode without forceobj=True, but has lifted loops.

File "../core/geometry.py", line 157:
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
    # first convert polygon to directed lines
    num_points_of_polygon = polygon.shape[1]
    ^

  state.func_ir.loc))
/home/svu/e0318980/.local/lib/python3.6/site-packages/numba/core/object_mode_passes.py:162: NumbaDeprecationWarning:
Fall-back from the nopython compilation path to the object mode compilation path has been detected, this is deprecated behaviour.

For more information visit https://numba.pydata.org/numba-doc/latest/reference/deprecation.html#deprecation-of-object-mode-fall-back              -behaviour-when-using-jit

File "../core/geometry.py", line 157:
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
    # first convert polygon to directed lines
    num_points_of_polygon = polygon.shape[1]
    ^

  state.func_ir.loc))
/home/svu/e0318980/.local/lib/python3.6/site-packages/numba/core/typed_passes.py:316: NumbaPerformanceWarning:
The keyword argument 'parallel=True' was specified but no transformation for parallel execution was possible.

To find out why, try turning on parallel diagnostics, see https://numba.pydata.org/numba-doc/latest/user/parallel.html#diagnostics f              or help.

File "../utils/eval.py", line 127:
@numba.jit(nopython=True,parallel=True)
def build_stage2_training(boxes, query_boxes, criterion, scores_3d, scores_2d, dis_to_lidar_3d,overlaps,tensor_index):
^

  state.func_ir.loc))
/home/svu/e0318980/.local/lib/python3.6/site-packages/numba/core/typed_passes.py:316: NumbaPerformanceWarning:
The keyword argument 'parallel=True' was specified but no transformation for parallel execution was possible.

To find out why, try turning on parallel diagnostics, see https://numba.pydata.org/numba-doc/latest/user/parallel.html#diagnostics f              or help.

File "../utils/eval.py", line 231:
@numba.jit(nopython=True, parallel=True)
def d3_box_overlap_kernel(boxes, qboxes, rinc, criterion=-1):
^

  state.func_ir.loc))
[100.0%][===================>][8.58it/s][09:38>00:00]
generate label finished(6.51/s). start eval:
validation_loss: 0.0
avg example to torch time: 1.877 ms
avg prep time: 31.322 ms
/home/svu/e0318980/.local/lib/python3.6/site-packages/numba/core/typed_passes.py:316: NumbaPerformanceWarning:
The keyword argument 'parallel=True' was specified but no transformation for parallel execution was possible.

To find out why, try turning on parallel diagnostics, see https://numba.pydata.org/numba-doc/latest/user/parallel.html#diagnostics f              or help.

File "../utils/eval.py", line 231:
@numba.jit(nopython=True, parallel=True)
def d3_box_overlap_kernel(boxes, qboxes, rinc, criterion=-1):
^

  state.func_ir.loc))
Car AP@0.70, 0.70, 0.70:
bbox AP:0.00, 0.00, 0.00
bev  AP:0.00, 0.00, 0.00
3d   AP:0.00, 0.00, 0.00
Car AP@0.70, 0.50, 0.50:
bbox AP:0.00, 0.00, 0.00
bev  AP:0.00, 0.00, 0.00
3d   AP:0.00, 0.00, 0.00

Car AP@0.70, 0.70, 0.70:
bbox AP:0.00, 0.00, 0.00
bev  AP:0.00, 0.00, 0.00
3d   AP:0.00, 0.00, 0.00
Car AP@0.70, 0.50, 0.50:
bbox AP:0.00, 0.00, 0.00
bev  AP:0.00, 0.00, 0.00
3d   AP:0.00, 0.00, 0.00

```

harris-mohamed, 2021-03-01:
I am observing the same behavior. Where should the highlighted changes be made?

pangsu0613, 2021-03-02:
@harris-mohamed , may I know which behavior you are observing?

harris-mohamed, 2021-03-02:
> [image]
> 
> Had this problem after I train, not sure why does the AP are 0

The same issue as here, but my AP have actual values. I tried changing the line from get_coco_eval_results to get_official_eval_result_v2, but it says it does not exist.

pangsu0613, 2021-03-02:
The first error occures because SECOND-V1.5 is written in older version numpy, and there are some changes in newer numpy versions that does not support that. But we don't need COCO results, we only care about KITTI evaluation results. So, the easiest way is to comment or delete 'result = get_coco_eval_result(gt_annos, dt_annos, class_names)' and the 'result' variable after that. You don't need to call 'get_official_eval_result_v2' because 'get_official_eval_result' is used to generate the KITTI result (you can find in line 22, only 'get_official_eval_result' are imported from second.utils.eval). Anyway, never mind, just comment or delete 'result = get_coco_eval_result(gt_annos, dt_annos, class_names)' and the 'result' variable after that, then you are good to go. I see your evaluation results, that's good.

closed 2021-03-03 by pangsu0613