pangsu0613/CLOCs #18: Inference on Kitti dataset

shaunkheng97, 2021-03-08:
May I know how can I run the inference on CLOCs? Do I need to write my own inference code?

pangsu0613, 2021-03-08:
Hello @shaunkheng97 , The Evaluation section in readme illustrates how to run inference using CLOCs.

shaunkheng97, 2021-03-09:
Is there a way to visualize my inference results on test images on 2d images or 3d point cloud?

pangsu0613, 2021-03-09:
The outputs of CLOCs are in the same format as SECOND-V1.5, so you could use the KITTI viewer under /CLOCs/second/kittiviewer, the instructions can be found in this page, https://github.com/traveller59/second.pytorch/tree/v1.5. kittiviewer could visualize the results on 3d point cloud and images. Another option is to use the MATLAB script from official kitti object development kit, this script can visualize 2D bbox and projected 3d bbox on images. The MATLAB script is original developed to visualize kitti labels (txt files), so you need to make some small changes to use it.

closed 2021-03-24 by pangsu0613