pangsu0613/CLOCs #23: How to quantize

harris-mohamed, 2021-03-25:
How would I go about quantizing the pretrained model for execution on an accelerator? I'm guessing I would quantize the 2D model, then the 3D model, and then finally the fusion layer.

pangsu0613, 2021-03-26:
Could you give more explanation about what is quantization and accelerator? I am not very familiar with these terms.

harris-mohamed, 2021-03-26:
My aim with CLOCs is to get it running on a small, indoor robot. The focus of my research is to be able to run CLOCs on a lightweight device, for which I will be using an edge TPU, a very tiny computer which is designed to run inference on models very quickly. PyTorch and Tensorflow support model quantization, which is a method of taking floating point weights and converting them to integer representations to save on model size and inference time without losing too much in accuracy. https://pytorch.org/docs/stable/quantization.html explains it pretty well.

pangsu0613, 2021-03-28:
Ok, I see, then I think you are right. First quantize the 2D and 3D model, then the CLOCs fusion layer.

closed 2021-05-08 by pangsu0613