pangsu0613/CLOCs #25: Should velodyne_reduced folder be empty?

arxidinakbar, 2021-04-20:
Hi, I've follow your instructions, and let the velodyne_reduced folder empty. But got this error when running train.py
Want some hints, thank you
```
Original Traceback (most recent call last):
  File "/opt/conda/envs/spconv-1.0/lib/python3.6/site-packages/torch/utils/data/_utils/worker.py", line 178, in _worker_loop
    data = fetcher.fetch(index)
  File "/opt/conda/envs/spconv-1.0/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/opt/conda/envs/spconv-1.0/lib/python3.6/site-packages/torch/utils/data/_utils/fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/workspace/aixiding/CLOCs/second/pytorch/builder/input_reader_builder.py", line 18, in __getitem__
    return self._dataset[idx]
  File "/workspace/aixiding/CLOCs/second/data/dataset.py", line 70, in __getitem__
    prep_func=self._prep_func)
  File "/workspace/aixiding/CLOCs/second/data/preprocess.py", line 313, in _read_and_prep_v9
    count=-1).reshape([-1, num_point_features])
FileNotFoundError: [Errno 2] No such file or directory: '/workspace/aixiding/CLOCs/KITTI_DATASET_ROOT/training/velodyne_reduced/002445.bin'
```

Best luck

pangsu0613, 2021-04-20:
Hello @arxidinakbar, leave that folder empty is just the first step, you still need to generate these reduced point clouds or simply download from here: (https://drive.google.com/drive/folders/1ScFUWPwzK5_VXb-LYQZuZVkiBj-dTMJ9?usp=sharing), I also put this link in the instruction.

arxidinakbar, 2021-04-20:
Hi @pangsu0613, thank you so much for quick reply.
You made it much clearer, and I may not able to download from the Google drive. So, if I generate reduced point cloud myself which command line should I run? Is it  'create_data.py' in second folder?

Best luck

pangsu0613, 2021-04-20:
@arxidinakbar, Yes, you could follow the directions here (https://github.com/traveller59/second.pytorch/tree/v1.5), "create reduced point cloud" subsession, if there are some weird errors, you may need to download the original SECOND V1.5 codebase and run that command (NOTE, It has to be v-1.5). Because I may have changed some things but I can not recall, sorry about that.
BTW, if Google Drive is not available to you, which cloud storage service is available for you?

arxidinakbar, 2021-04-20:
@pangsu0613, Ah I see, then I could understand whole pre and post steps now.
For now it would be great if there is Baidu drive link for the dataset.
Again, thank you for all efforts done here.

Best luck

pangsu0613, 2021-04-20:
@arxidinakbar , I see, but currently my Internet provider has partially blocked Baidu services, the connection to their server is not stable. I'll try sometime later.

arxidinakbar, 2021-04-20:
@pangsu0613 What a pity. I would try to generate new reduced velodyne myself, since I know the steps.

Best luck

closed 2021-04-20 by pangsu0613