pangsu0613/CLOCs #26: Fusion of other 3D and 2D detectors

kshitij3112, 2021-04-20:
Hi. Congratulations on your work! I have followed the instructions and am able to reproduce the results. 
One thing which remains unclear is step 2 of "Fusion of other 2D and 3D detectors". In which file or path to save the 3D detector outputs?
Also, after making the required changes as suggested in steps 2 and 3, should I just run the same command for the training?

pangsu0613, 2021-04-20:
Hello @kshitij3112, thank you for your interest in CLOCs! 
I understand these steps are quite complicated. For "fusion of other 2D and 3D detectors", one needs to save the 3D detector output into files (such as .pkl), and read these files during training. This reading process should replace the original SECOND inference process in CLOCs. So, there are some works to do here, not just modify the path. 
I will have another lab mate helping me maintain this repo in the end of April and we will work on making this whole process clearer and simpler. We will let you know when we have some thing new. Thank you.

kshitij3112, 2021-04-28:
Thank you for your reply. Eagerly waiting for that. :)

WuYiYiYi, 2021-07-29:
Hi，congratulate your work ! I also want know about "Fusion of other 2D and 3D detectors", can you give me some sugguestions ? Will you realease this code ?

1553545365, 2021-12-14:
Thank you for your great work! I am waiting for your update too. I want to fusion of other 3D and 2D detectors @pangsu0613  Thank you a lot.

KuntimaKiala, 2022-02-09:
Hi, Congrats for this amazing Work, I have a doubt concerning step 3, first why do you need to modify the corresponding parameters in the CLOCs code to adjust the number 2D/3D candidates ? Is it to reduce the size of the tensor ? and finally, where in the code you modify  it ? in train_stage_2 function ?. Thanks