pangsu0613/CLOCs #27: no module named 'fusion'

arxidinakbar, 2021-04-20:
Hi, When I running train.py got another error as below:

```
Traceback (most recent call last):
  File "./pytorch/train.py", line 26, in <module>
    from second.pytorch.models import fusion
ImportError: cannot import name 'fusion'
```
'fusion.py' did exist in models folder, and I did add the CLOCs to the pythonpath.

Best luck

pangsu0613, 2021-04-20:
Maybe because you have multiple SECOND in your $PYTHONPATH, this could make the system confused, make sure you only have CLOCs in your pythonpath and remove other SECOND (such as SECOND-v1.5 or v1.6) from your $PYTHONPATH.

arxidinakbar, 2021-04-21:
Hi, @pangsu0613 It is the case which you mentioned.
But after remove the SECOND pythonpath, I still met this error (I did source ~/.bashrc)

Best luck

pangsu0613, 2021-04-22:
You could send me your ~/.bashrc (hide any personal info that you don't want me to see), and the result for running "echo $PYTHONPATH" in your terminal, and I could have a look.

arxidinakbar, 2021-04-22:
Hi, thank you for generous helping!
I've used remote server to training model. Somehow the `source ~/.bashrc` command did'nt work led to the error.
I found original SECOND pythonpath in echo $PYTHONPATH return which bring the error.
Now I could running train.py, all problem solved.

Best luck

closed 2021-04-22 by arxidinakbar