pangsu0613/CLOCs #28: How to evaluate the performance of the model according to the segmentation distance on Kitti?

jlqzzz, 2021-04-23:
How to evaluate the performance of the model according to the segmentation distance on Kitti? Such as 0-30m，30-50，50-70.4.
Where should I change the evaluation code to control the evaluation metric ?

pangsu0613, 2021-04-24:
Hello, you need to modify two places in the "clean_data" function in /CLOCs/second/utils/eval.py. 
Say if you want to evaluate 0-30m.
Line 66: add another condition, "gt_anno["location"][i,0]**2 + gt_anno["location"][i,2]**2 > 900", this means ground truth boxes that are 30 meters away are IGNORED.
Line 87: add another similar condition, dt_anno["location"][i,0]**2 + dt_anno["location"][i,2]**2 > 900, this means predicted boxes that are 30 meters away are IGNORED.

jlqzzz, 2021-04-25:
> Hello, you need to modify two places in the "clean_data" function in /CLOCs/second/utils/eval.py.
> Say if you want to evaluate 0-30m.
> Line 66: add another condition, "gt_anno["location"][i,0]**2 + [...]

Thank you very much for your reply!
Please confirm my understanding again.

1. For example, 30-50 should correspond to "gt_anno["location"][i,0]**2 + gt_anno["location"][i,2]**2 < 900 | gt_anno["location"][i,0]**2 + gt_anno["location"][i,2]**2 > 2500" ？
That is,
if ( (gt_anno["occluded"][i] > MAX_OCCLUSION[difficulty])
                or (gt_anno["truncated"][i] > MAX_TRUNCATION[difficulty])
                or (height <= MIN_HEIGHT[difficulty]) 
                or (gt_anno["location"][i,0]**2 + gt_anno["location"][i,2]**2 < 900 | gt_anno["location"][i,0]**2 + gt_anno["location"][i,2]**2 > 2500) ):
            # if gt_anno["difficulty"][i] > difficulty or gt_anno["difficulty"][i] == -1:
            ignore = True

2.         if ( (height < MIN_HEIGHT[difficulty])
                or (dt_anno["location"][i,0]**2 + dt_anno["location"][i,2]**2 < 900 | dt_anno["location"][i,0]**2 + dt_anno["location"][i,2]**2 > 2500) ):
            ignored_dt.append(1)
        elif valid_class == 1:
            ignored_dt.append(0)
        else:
            ignored_dt.append(-1)

For 1, I think it should be right, for 2. I am not sure. Please confirm it for me, thank you!

jlqzzz, 2021-04-25:
@pangsu0613

pangsu0613, 2021-04-25:
Hello @jlqzzz , yes, I think you are right.

closed 2021-05-08 by pangsu0613