pangsu0613/CLOCs #35: AssertionError with latest spconv version

Jamie725, 2021-06-19:
As title. Could you please help me with this issue? Thanks!!!

pangsu0613, 2021-06-19:
Hi @Jamie725 , you need to use the spconv with the commit that I mentioned in the readme. This is constraint by SECOND-V1.5.

Jamie725, 2021-06-21:
Hi @pangsu0613,
Thanks for the information. I followed the following commands but failed building spconv:

git clone https://github.com/traveller59/spconv.git
cd spconv
git checkout 8da6f967fb9a054d8870c3515b1b44eca2103634

I got building errors with: 
error: no matching function for call to ‘torch::jit::RegisterOperators::RegisterOperators(const char [28], <unresolved overloaded function type>)’
   20 |     torch::jit::RegisterOperators("spconv::get_indice_pairs_2d", &spconv::getIndicePair<2>)

[image]

Therefore, I was wondering if I did anything wrong. Could you please share with me the some suggested instructions to install this specific commit on spconv?

Thank you very much for your help!

pangsu0613, 2021-06-21:
Hello @Jamie725 , I understand installing spconv is not easy. You can try change torch::jit::RegisterOperators into torch::RegisterOperators() under /spconv/src/spconv/all.cc.
You can also check the issues section in spconv repo for other instructions.

Jamie725, 2021-06-21:
Hi @pangsu0613,
It works! Thank you very much for your help!

closed 2021-06-21 by Jamie725

Anirbanbhk88, 2022-04-03:
@Jamie725 how did you solve the above issue