pangsu0613/CLOCs #36: Bug in fusion process

Treemann, 2021-06-21:
Hi Pang,

Thanks for your contribution~

I notice that there is a bug in your code: [voxelnet.py](https://github.com/pangsu0613/CLOCs/blob/master/second/pytorch/models/voxelnet.py) L498-L501.
``` python
            box_2d_detector = np.zeros((200, 4))
            box_2d_detector[0:top_predictions.shape[0],:]=top_predictions[:,:4]
            box_2d_detector = top_predictions[:,:4]
            box_2d_scores = top_predictions[:,4].reshape(-1,1)
```
Actually, the shape of "box_2d_detector" is changeable (instead of remaining 200) with your implementation.
Thus assigning values to "out_1" according to the coordinates in (fusion)[https://github.com/pangsu0613/CLOCs/blob/master/second/pytorch/models/fusion.py] L48-L50 is wrong logically.
But due to the maxpooling operation along the axis of box_2d, I think this bug has no effect on the results.
I think we could replace the aforementioned code with:
``` python
            box_2d_detection = np.zeros((200, 5)) 
            box_2d_detection[0:top_predictions.shape[0],:]=top_predictions[:,:5]
            box_2d_detector = box_2d_detection[:,:4]
            box_2d_scores = box_2d_detection[:,4].reshape(-1,1)
```
Or pass the shape to fusion_layer.forward()...

pangsu0613, 2021-06-21:
Hello @Treemann, thank you for your interests in CLOCs and pointing this out. There is some legacy code there, a while ago, 200 is set as the maximum number of 2D detections. You are right, the shape of box_2d_detector varies at each iter, but it is smaller than 200, so orginally we only fill out the number of 2D detections and leave the rest as empty, these empty ones does not have any impact on the fusion and also won't affect the indices in the original sparse input tensor. Anyway, this issue currenly does not affect the fusion. We will clean and fix this in the next update and thank you very much for pointing this out.

Treemann, 2021-06-21:
Hi Pang,

Thanks for your reply~ I agree with your explanation. 

And about the  indices in the original sparse input tensor, I think the shape of `box_2d_detector` passed to `build_stage2_training` varies, then K varies each time (`K = query_boxes.shape[0]`), which means the index `tensor_index[ind,0] = k` is related to the shape of K instead of 200. Then the code in fusion.py assigns values to out_1 (dim1=200) according to the indices which are calculated related to the shape of K, I mean actually there are some values placed on the wrong position along dim1 during  the reconstruction of out_1. But since the maxpool is conducted along dim1, then the results remain good although the reconstruction of out_1 has some problem on dim1.
``` python
out_1 = torch.zeros(1,200,70400,dtype = input_1.dtype,device = input_1.device)
out_1[:,:,:] = -9999999
out_1[:,tensor_index[:,0],tensor_index[:,1]] = x[0,:,0,:]
```

Thanks!

closed 2021-06-22 by Treemann