pangsu0613/CLOCs #37: Cuda out of memory

Yaonian72, 2021-06-25:
When I run 

python ./pytorch/train.py train --config_path=./configs/car.fhd.config --model_dir=/home/yao/clocs/model_dir/

Environment
Ubuntu20.04
GTX3080
Conda environment:
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
blas                      1.0                         mkl  
ca-certificates           2021.5.30            ha878542_0    conda-forge
certifi                   2021.5.30        py37h89c1867_0    conda-forge
cudatoolkit               11.0.221             h6bb024c_0    anaconda
cycler                    0.10.0                   pypi_0    pypi
dataclasses               0.6                      pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
easydict                  1.9                      pypi_0    pypi
fire                      0.4.0                    pypi_0    pypi
freetype                  2.10.4               h0708190_1    conda-forge
future                    0.18.2                   pypi_0    pypi
imageio                   2.9.0                    pypi_0    pypi
intel-openmp              2021.2.0           h06a4308_610  
jpeg                      9b                   h024ee3a_2  
kiwisolver                1.3.1                    pypi_0    pypi
kornia                    0.5.4                    pypi_0    pypi
lcms2                     2.12                 h3be6417_0  
ld_impl_linux-64          2.35.1               hea4e1c9_2    conda-forge
libffi                    3.3                  h58526e2_2    conda-forge
libgcc-ng                 9.3.0               h2828fa1_19    conda-forge
libgomp                   9.3.0               h2828fa1_19    conda-forge
libpng                    1.6.37               h21135ba_2    conda-forge
libstdcxx-ng              9.3.0               h6de172a_19    conda-forge
libtiff                   4.2.0                h85742a9_0  
libuv                     1.41.0               h7f98852_0    conda-forge
libwebp-base              1.2.0                h7f98852_2    conda-forge
llvmlite                  0.36.0                   pypi_0    pypi
lz4-c                     1.9.3                h9c3ff4c_0    conda-forge
matplotlib                3.4.2                    pypi_0    pypi
mkl                       2021.2.0           h06a4308_296  
mkl-service               2.4.0            py37h5e8e339_0    conda-forge
mkl_fft                   1.3.0            py37h42c9631_2  
mkl_random                1.2.2            py37h219a48f_0    conda-forge
ncurses                   6.2                  h58526e2_4    conda-forge
networkx                  2.5.1                    pypi_0    pypi
ninja                     1.10.2               h4bd325d_0    conda-forge
numba                     0.53.1                   pypi_0    pypi
numpy                     1.21.0                   pypi_0    pypi
numpy-base                1.20.2           py37hfae3a4d_0  
olefile                   0.46               pyh9f0ad1d_1    conda-forge
opencv-python             4.5.2.54                 pypi_0    pypi
openssl                   1.1.1k               h7f98852_0    conda-forge
pcdet                     0.3.0+65bd4cd             dev_0    <develop>
pillow                    8.2.0            py37he98fc37_0  
pip                       21.1.2             pyhd8ed1ab_0    conda-forge
protobuf                  3.17.3                   pypi_0    pypi
pybind11                  2.6.2                    pypi_0    pypi
pyparsing                 2.4.7                    pypi_0    pypi
python                    3.7.10          hffdb5ce_100_cpython    conda-forge
python-dateutil           2.8.1                    pypi_0    pypi
python_abi                3.7                     1_cp37m    conda-forge
pywavelets                1.1.1                    pypi_0    pypi
pyyaml                    5.4.1                    pypi_0    pypi
readline                  8.1                  h46c0cb4_0    conda-forge
scikit-image              0.18.1                   pypi_0    pypi
scipy                     1.7.0                    pypi_0    pypi
setuptools                49.6.0           py37h89c1867_3    conda-forge
shapely                   1.7.1                    pypi_0    pypi
six                       1.16.0             pyh6c4a22f_0    conda-forge
[... 261 lines cut ...]
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/fire/core.py", line 681, in _CallAndUpdateTrace
    component = fn(*varargs, **kwargs)
  File "./pytorch/train.py", line 371, in train
    raise e
  File "./pytorch/train.py", line 240, in train
    all_3d_output_camera_dict, all_3d_output, top_predictions, fusion_input,tensor_index = net(example_torch,detection_2d_path)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/yao/clocs/second/pytorch/models/voxelnet.py", line 310, in forward
    preds_dict = self.rpn(spatial_features)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/yao/clocs/second/pytorch/models/rpn.py", line 312, in forward
    x = self.blocks[i](x)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/yao/clocs/torchplus/nn/modules/common.py", line 89, in forward
    input = module(input)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/nn/modules/batchnorm.py", line 178, in forward
    self.eps,
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/nn/functional.py", line 2282, in batch_norm
    input, weight, bias, running_mean, running_var, training, momentum, eps, torch.backends.cudnn.enabled
RuntimeError: CUDA out of memory. Tried to allocate 20.00 MiB (GPU 0; 9.78 GiB total capacity; 1.54 GiB already allocated; 36.44 MiB free; 1.57 GiB reserved in total by PyTorch)

pangsu0613, 2021-06-25:
Hello @Yaonian72, I have tested CLOCs on RTX 3080, RTX 3080 has 10GB GPU memory, which is sufficient for running CLOCs (we only run SECOND online, no 2D detectors running online). I believe there are other applications on your machine that takes too much GPU memory, I suggest close them or reboot then run CLOCs.

Yaonian72, 2021-06-25:
> Hello @Yaonian72, I have tested CLOCs on RTX 3080, RTX 3080 has 10GB GPU memory, which is sufficient for running CLOCs (we only run SECOND online, no 2D detectors running online). I believe there [...]

Thank you for replying promptly.
The Nvidia-smi show that no memory is occupied
Fri Jun 25 18:00:08 2021       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 460.80       Driver Version: 460.80       CUDA Version: 11.2     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce RTX 3080    Off  | 00000000:0B:00.0  On |                  N/A |
|  0%   38C    P8    26W / 320W |    280MiB / 10010MiB |      2%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1280      G   /usr/lib/xorg/Xorg                176MiB |
|    0   N/A  N/A      1583      G   /usr/bin/gnome-shell               32MiB |
|    0   N/A  N/A      6916      G   ...AAAAAAAAA= --shared-files       68MiB |
+-----------------------------------------------------------------------------+

However,after changing the batch size from 1 to 2, the issue disappears, but another issue raised
o/clocs/model_dir/
2d detection path: /home/yao/clocs/d2_detection_data
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from ../model_dir/voxelnet-30950.tckpt
Restoring parameters from /home/yao/clocs/model_dir/adam_optimizer-0.tckpt
{'Car': 5}
[-1]
load 14357 Car database infos
load 2207 Pedestrian database infos
load 734 Cyclist database infos
load 1297 Van database infos
load 56 Person_sitting database infos
load 488 Truck database infos
load 224 Tram database infos
load 337 Misc database infos
After filter database:
load 10520 Car database infos
load 2104 Pedestrian database infos
load 594 Cyclist database infos
load 826 Van database infos
load 53 Person_sitting database infos
load 321 Truck database infos
load 199 Tram database infos
load 259 Misc database infos
remain number of infos: 3712
remain number of infos: 3769
WORKER 0 seed: 1624657916
WORKER 1 seed: 1624657917
WORKER 2 seed: 1624657918
/home/yao/clocs/second/core/geometry.py:146: NumbaWarning: 
Compilation is falling back to object mode WITH looplifting enabled because Function "points_in_convex_polygon_jit" failed type inference due to: No implementation of function Function(<built-in function getitem>) found for signature:
 
 >>> getitem(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))
 
There are 22 candidate implementations:
   - Of which 20 did not match due to:
   Overload of function 'getitem': File: <numerous>: Line N/A.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    No match.
   - Of which 2 did not match due to:
[... 180 lines cut ...]
    component = fn(*varargs, **kwargs)
  File "./pytorch/train.py", line 371, in train
    raise e
  File "./pytorch/train.py", line 231, in train
    example = next(data_iter)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 521, in __next__
    data = self._next_data()
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 1203, in _next_data
    return self._process_data(data)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/utils/data/dataloader.py", line 1229, in _process_data
    data.reraise()
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/_utils.py", line 425, in reraise
    raise self.exc_type(msg)
ValueError: Caught ValueError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/utils/data/_utils/worker.py", line 287, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/torch/utils/data/_utils/fetch.py", line 47, in fetch
    return self.collate_fn(data)
  File "/home/yao/clocs/second/data/preprocess.py", line 53, in merge_second_batch
    ret[key] = np.stack(elems, axis=0)
  File "<__array_function__ internals>", line 6, in stack
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/numpy/core/shape_base.py", line 427, in stack
    raise ValueError('all input arrays must have the same shape')
ValueError: all input arrays must have the same shape

pangsu0613, 2021-06-25:
Current version of CLOCs only supports batch_size=1.
Are you using the same commit of spconv that I mentioned in the readme? I heard newer version of spconv requires much more memories.

Yaonian72, 2021-06-25:
7342772

> Current version of CLOCs only supports batch_size=1.
> Are you using the same commit of spconv that I mentioned in the readme? I heard newer version of spconv requires much more memories.

Thank you!
It works after changing the version of spconv. Before that, I used spconv1.2 and spconv1.0 7342772. They can not work.

Yaonian72, 2021-06-25:
> Current version of CLOCs only supports batch_size=1.
> Are you using the same commit of spconv that I mentioned in the readme? I heard newer version of spconv requires much more memories.

Another issue raised.It's that normal?
2d detection path: /home/yao/clocs/d2_detection_data
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from ../model_dir/voxelnet-30950.tckpt
Restoring parameters from /home/yao/clocs/model_dir/adam_optimizer-0.tckpt
{'Car': 5}
[-1]
load 14357 Car database infos
load 2207 Pedestrian database infos
load 734 Cyclist database infos
load 1297 Van database infos
load 56 Person_sitting database infos
load 488 Truck database infos
load 224 Tram database infos
load 337 Misc database infos
After filter database:
load 10520 Car database infos
load 2104 Pedestrian database infos
load 594 Cyclist database infos
load 826 Van database infos
load 53 Person_sitting database infos
load 321 Truck database infos
load 199 Tram database infos
load 259 Misc database infos
remain number of infos: 3712
remain number of infos: 3769
WORKER 0 seed: 1624659846
WORKER 1 seed: 1624659847
WORKER 2 seed: 1624659848
/home/yao/clocs/second/core/geometry.py:146: NumbaWarning: 
Compilation is falling back to object mode WITH looplifting enabled because Function "points_in_convex_polygon_jit" failed type inference due to: No implementation of function Function(<built-in function getitem>) found for signature:
 
 >>> getitem(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))
 
There are 22 candidate implementations:
   - Of which 20 did not match due to:
   Overload of function 'getitem': File: <numerous>: Line N/A.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    No match.
   - Of which 2 did not match due to:
   Overload in function 'GetItemBuffer.generic': File: numba/core/typing/arraydecl.py: Line 162.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    Rejected as the implementation raised a specific error:
      TypeError: unsupported array index type list(int64)<iv=None> in Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>)
  raised from /home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/numba/core/typing/arraydecl.py:69

During: typing of intrinsic-call at /home/yao/clocs/second/core/geometry.py (162)

File "core/geometry.py", line 162:
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
        vec1 = polygon - polygon[:, [num_points_of_polygon - 1] +
                                 list(range(num_points_of_polygon - 1)), :]
                                 ^

  @numba.jit
/home/yao/clocs/second/core/geometry.py:146: NumbaWarning: 
Compilation is falling back to object mode WITH looplifting enabled because Function "points_in_convex_polygon_jit" failed type inference due to: No implementation of function Function(<built-in function getitem>) found for signature:
 
 >>> getitem(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))
 
There are 22 candidate implementations:
   - Of which 20 did not match due to:
   Overload of function 'getitem': File: <numerous>: Line N/A.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
[... 425 lines cut ...]
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
    # first convert polygon to directed lines
    num_points_of_polygon = polygon.shape[1]
    ^

  state.func_ir.loc))
Traceback (most recent call last):
  File "./pytorch/train.py", line 918, in <module>
    fire.Fire()
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/fire/core.py", line 141, in Fire
    component_trace = _Fire(component, args, parsed_flag_args, context, name)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/fire/core.py", line 471, in _Fire
    target=component.__name__)
  File "/home/yao/Documents/anaconda3/envs/3d/lib/python3.7/site-packages/fire/core.py", line 681, in _CallAndUpdateTrace
    component = fn(*varargs, **kwargs)
  File "./pytorch/train.py", line 371, in train
    raise e
  File "./pytorch/train.py", line 329, in train
    model_cfg.lidar_input)
  File "./pytorch/train.py", line 467, in predict_kitti_to_anno
    predictions_dicts = predict_v2(net,example, all_3d_output)
  File "./pytorch/train.py", line 871, in predict_v2
    opp_labels = (box_preds[..., -1] > 0) ^ dir_labels.byte()
RuntimeError: result type Byte can't be cast to the desired output type Bool

pangsu0613, 2021-06-25:
Refer here: https://github.com/pangsu0613/CLOCs#common-errors--solutions

Yaonian72, 2021-06-25:
> Refer here: https://github.com/pangsu0613/CLOCs#common-errors--solutions

OK I saw it.
BTW, do you think is it possible to use clocs architecture to fuse centerfusion and 3dssd ,and then trained by nuscene dataset?

Yaonian72, 2021-06-25:
> Refer here: https://github.com/pangsu0613/CLOCs#common-errors--solutions

It is there any recommendation for the torch version? Because sometimes GeForce RTX 3080 with CUDA capability sm_86 is not compatible with the PyTorch version

pangsu0613, 2021-06-25:
https://github.com/pangsu0613/CLOCs/issues/33#issuecomment-860574933, this guy test CenterPoint with CLOCs.
Yes, CLOCs works for any 2D and 3D detector pairs. CenterFusion and 3DSSD are both 3D detectors, I am not sure how you would like to use them, using CLOCs to fuse two 3D detectors is possible but I have not tested. You need to change 2D IoU association into 3D IoU. And yes, you can definitely use nuScenes, but need modifications.
On my RTX 3080, I use python3.7 and pytorch 1.7.1.

Yaonian72, 2021-06-26:
> [#33 (comment)](https://github.com/pangsu0613/CLOCs/issues/33#issuecomment-860574933), this guy test CenterPoint with CLOCs.
> Yes, CLOCs works for any 2D and 3D detector pairs. CenterFusion and 3DSSD are both 3D detectors, I am not sure how you would like to use them, using CLOCs to fuse two 3D detectors is possible but I have not tested. You need to change 2D IoU association into 3D IoU. And yes, you can definitely use nuScenes, but need modifications.
> On my RTX 3080, I use python3.7 and pytorch 1.7.1.

Thank you for telling me this, I will read it carefully.
Sorry may I ask your Cuda version,I try several times to install 1.7.1 and1.7.0, it still appears the same issue.

pangsu0613, 2021-06-26:
I have CUDA 11.0.221, I did it in a conda environment.

closed 2021-06-29 by pangsu0613

zijianh4, 2022-07-29:
> Current version of CLOCs only supports batch_size=1.
> Are you using the same commit of spconv that I mentioned in the readme? I heard newer version of spconv requires much more memories.

Is there any plan to update the current implementation to support batch_size > 1? Thanks!