pangsu0613/CLOCs #38: TypeError: 'numpy.float64' object cannot be interpreted as an integer

urbansound8K, 2021-06-27:
Traceback (most recent call last):
  File "./pytorch/train.py", line 918, in <module>
    fire.Fire()
  File "/home/cute/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/fire/core.py", line 141, in Fire
    component_trace = _Fire(component, args, parsed_flag_args, context, name)
  File "/home/cute/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/fire/core.py", line 471, in _Fire
    target=component.__name__)
  File "/home/cute/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/fire/core.py", line 681, in _CallAndUpdateTrace
    component = fn(*varargs, **kwargs)
  File "./pytorch/train.py", line 371, in train
    raise e
  File "./pytorch/train.py", line 356, in train
    result = get_coco_eval_result(gt_annos, dt_annos, class_names)
  File "/media/cute/New Volume/AAA/YYY/fusion/ClOCss/second/utils/eval.py", line 1017, in get_coco_eval_result
    gt_annos, dt_annos, current_classes, overlap_ranges, compute_aos)
  File "/media/cute/New Volume/AAA/YYY/fusion/ClOCss/second/utils/eval.py", line 828, in do_coco_style_eval
    min_overlaps[:, i, j] = np.linspace(*overlap_ranges[:, i, j])
  File "<__array_function__ internals>", line 6, in linspace
  File "/home/cute/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/numpy/core/function_base.py", line 113, in linspace
    num = operator.index(num)
TypeError: 'numpy.float64' object cannot be interpreted as an integer

why am I getting this, please?

and when I trained I kept the velodyne_reduced with veldoyne data and velodyne is empty! it did not work with the opposite!!!

pangsu0613, 2021-06-27:
Hello @urbansound8K , For the first question, refer https://github.com/pangsu0613/CLOCs/issues/16, the main problem is numpy versions, but since we don't need coco result, remove it is the simplest solution.
For the second question, follow https://github.com/pangsu0613/CLOCs#prepare-dataset-kitti carefully, note that "velodyne" folder is for original KITTI velodyne point cloud, for "velodyne_reduced" is the one that actually used by SECOND-V1.5, it contains the downsampled point clouds, you need to download these reduced point clouds from the link that I put in https://github.com/pangsu0613/CLOCs#prepare-dataset-kitti.

urbansound8K, 2021-06-29:
could you tell how did you make the downsampled point clouds files, please?

pangsu0613, 2021-06-29:
Hi @urbansound8K , https://github.com/traveller59/second.pytorch/tree/v1.5#prepare-dataset, in "create reduced point cloud" subsection, it tells you how to generate the reduced point cloud, basically, it removes the points out of view, because KITTI only have cameras facing front, while LiDAR has 360 degrees field of view.
During training CLOCs, I run SECOND in real time, I did this just because it is simple (and SECOND is pretty fast). If you want, you can run SECOND and save the results for CLOCs training, then you don't need to run SECOND online, but you need to modify the codebase, we are also working on this now, will push new versions when it is available. https://github.com/pangsu0613/CLOCs#evaluation, this section is about how to run inference. I am not sure if this answers your question, feel free to ping me if you have further questions.

FolliesHandle, 2021-07-02:
@urbansound8K Additionally, if you would still like to see the coco results, if you modify line 113 of /home/$USER/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/numpy/core/function_base.py to be `num = operator.index(int(num))` you should be able to see those results

urbansound8K, 2021-07-06:
Hello @pangsu0613 

could it be possible for you to do me a favor, please?

I trained the model and I followed the preperation you provided in the README, but still got wired results!!!

Predict_test:  False
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
load existing model for fusion layer
Restoring parameters from CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
/media/cute/New Volume/AAA/YYY/fusion/ClOCss/second/core/geometry.py:146: NumbaWarning: 
Compilation is falling back to object mode WITH looplifting enabled because Function "points_in_convex_polygon_jit" failed type inference due to: No implementation of function Function(<built-in function getitem>) found for signature:
 
 >>> getitem(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))
 
There are 22 candidate implementations:
   - Of which 20 did not match due to:
   Overload of function 'getitem': File: <numerous>: Line N/A.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    No match.
   - Of which 2 did not match due to:
   Overload in function 'GetItemBuffer.generic': File: numba/core/typing/arraydecl.py: Line 162.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    Rejected as the implementation raised a specific error:
      TypeError: unsupported array index type list(int64)<iv=None> in Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>)
  raised from /home/cute/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/numba/core/typing/arraydecl.py:69

During: typing of intrinsic-call at /media/cute/New Volume/AAA/YYY/fusion/ClOCss/second/core/geometry.py (162)

File "core/geometry.py", line 162:
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
        vec1 = polygon - polygon[:, [num_points_of_polygon - 1] +
                                 list(range(num_points_of_polygon - 1)), :]
                                 ^

  @numba.jit
/media/cute/New Volume/AAA/YYY/fusion/ClOCss/second/core/geometry.py:146: NumbaWarning: 
Compilation is falling back to object mode WITHOUT looplifting enabled because Function "points_in_convex_polygon_jit" failed type inference due to: Cannot determine Numba type of <class 'numba.core.dispatcher.LiftedLoop'>

File "core/geometry.py", line 170:
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
    cross = 0.0
    for i in range(num_points):
    ^

  @numba.jit
/home/cute/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/numba/core/object_mode_passes.py:152: NumbaWarning: Function "points_in_convex_polygon_jit" was compiled in object mode without forceobj=True, but has lifted loops.

File "core/geometry.py", line 157:
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
    # first convert polygon to directed lines
    num_points_of_polygon = polygon.shape[1]
    ^

  state.func_ir.loc))
/home/cute/anaconda3/envs/spconv-1.0/lib/python3.6/site-packages/numba/core/object_mode_passes.py:162: NumbaDeprecationWarning: 
Fall-back from the nopython compilation path to the object mode compilation path has been detected, this is deprecated behaviour.

For more information visit https://numba.pydata.org/numba-doc/latest/reference/deprecation.html#deprecation-of-object-mode-fall-back-behaviour-when-using-jit

File "core/geometry.py", line 157:
def points_in_convex_polygon_jit(points, polygon, clockwise=True):
    <source elided>
[... 39 lines cut ...]
def d3_box_overlap_kernel(boxes, qboxes, rinc, criterion=-1):
^

  state.func_ir.loc))
Car AP@0.70, 0.70, 0.70:
bbox AP:38.95, 33.66, 29.18
bev  AP:2.89, 2.67, 2.12
3d   AP:0.97, 0.79, 0.65
aos  AP:22.53, 19.04, 16.56
Car AP@0.70, 0.50, 0.50:
bbox AP:38.95, 33.66, 29.18
bev  AP:12.86, 11.50, 9.78
3d   AP:9.28, 8.18, 6.81
aos  AP:22.53, 19.04, 16.56

Car AP@0.70, 0.70, 0.70:
bbox AP:38.95, 33.66, 29.18
bev  AP:2.89, 2.67, 2.12
3d   AP:0.97, 0.79, 0.65
aos  AP:22.53, 19.04, 16.56
Car AP@0.70, 0.50, 0.50:
bbox AP:38.95, 33.66, 29.18
bev  AP:12.86, 11.50, 9.78
3d   AP:9.28, 8.18, 6.81
aos  AP:22.53, 19.04, 16.56

pangsu0613, 2021-07-06:
Hi @urbansound8K, you missed https://github.com/pangsu0613/CLOCs#preparation, step 2, you also need to download SECOND-V1.5 pretrained model from the link that I put in the README, and put it under the specified directory. Note that you need two pretrained models, one for SECOND-V1.5, the other for CLOCs, this is because in the current codebase, we are still running SECOND on the fly.
BTW, you can follow https://github.com/pangsu0613/CLOCs#common-errors--solutions, the second point, to hide these unrelated numba warnings, that would make your output cleaner.

closed 2021-07-18 by pangsu0613