pangsu0613/CLOCs #39: Running Own 2D detector

arsalan311, 2021-07-16:
Hello, I am running my own 2D detector, but I do not understand the Kitti Fusion format in which the sigmoid data is to be saved. 
And are the sigmoid data generated only on Test Data? since the mentioned folder has 7840 files.
Sorry I am new to this If I missed anything.

pangsu0613, 2021-07-16:
Hi @arsalan311, if you are using your own 2D detector, you need to generate 2D detections in KITTI format as I did. The sigmoid score is the confidence score for each detection from the detector, it is a float number between 0.0 to 1.0. This is because for most neural network based detectors, the raw score output from the network is something similar to a logarithm score (probably range between around -10 to 10), in order to map them into probabilities (the final confidence score is like the probability of how likely it is the target object), these logarithm scores need to go through a sigmoid function, then the output score is the sigmoid score in range 0.0 to 1.0. Some networks include this sigmoid function into the model network as a sigmoid layer in the output layer, some network put it outside of the network, but meanings are the same. 
There are 7481 frames in KITTI training set, 7518 frame in KITTI test set. No ground truth labels are provided for test set, you need to generate results and submit to KITTI for evaluation on test set. So most researchers divide the training set into a mini-training set that contains 3712 frames, and a mini-validation set that contains 3769 frames, and we can do evaluation using the mini-validation set locally.
Let me know if you have further quesitons.

closed 2021-07-21 by pangsu0613

arsalan311, 2021-08-11:
Hi, Thanks for the Reply. I ran a different 2D detector (YOLO3) on Kitti
dataset and saved the sigmoid values in Kitti format FOR fusion as
mentioned (as mentioned I did not change the  Confidence threshold  but
changed NMS thresh to 0.0 ).  I saved the notepad files and pasted in the "
d2_detection_data" folder.
But I am getting zero accuracy when I train the Fusion. Do I have to update
any other file like pkl? or just pasting the text files should work?