pangsu0613/CLOCs #41: Implementation of Cascade RCNN used

abhishek-t-naive, 2021-07-20:
Hi,
Thank you for sharing your work with us.

For my experiments, I want to modify the Camera (2D Detections) branch and LiDAR (3D Detections) branch before I run them through the CLOCs fusion network.

1. For the LiDAR branch, I can do it easily by modifying the pre-trained weights provided [here](https://drive.google.com/drive/folders/1ScFUWPwzK5_VXb-LYQZuZVkiBj-dTMJ9?usp=sharing).

2. But for the Camera branch, the current codebase is using the bounding boxes provided [here](https://drive.google.com/drive/folders/1ScFUWPwzK5_VXb-LYQZuZVkiBj-dTMJ9?usp=sharing). There are no pre-trained weights available (which I am guessing stems from the fact that the code doesn't use pre-trained weights, instead uses the bounding boxes)

So, I have the option of producing these bounding boxes with modified weights for which I would like to know which Cascade RCNN implementation(Including the backbone network) was used to produce the current given set of bounding boxes. It would be great if you can let me know about the same.

pangsu0613, 2021-07-20:
Hello @abhishek-t-naive, thank you for your interests in CLOCs, the cascade-RCNN that I am using is from this repo: https://github.com/zhaoweicai/mscnn, it is a pretty old one, because when I developed CLOCs, which is nearly 3 years ago, there are very limited opensource 2D detectors that perform good in KITTI. But for now, you can use your own 2D detector to generate 2D detections in KITTI format and feed them to CLOCs.

abhishek-t-naive, 2021-07-20:
Thank you for your quick response.

I will try to use my own 2D detector. I wanted to ask if it is possible to change the current CLOCs codebase to use a 2D detector network inbuilt instead of using the box produced by a 2d detector?

Or you think that is going to be difficult to implement?

pangsu0613, 2021-07-20:
It is possible and I don't think it is difficult to implement. My concern is, from hardware perspective, running a 3D detector and a heavy 2D detector simultatneously on one standard PC (single GPU platform) is difficult, you would encounter out of GPU memory issue. But if you have multi-GPU workstation, that's another story.

abhishek-t-naive, 2021-07-20:
That is a valid concern actually. I think I have a machine which has multiple-GPUs so I can probably give it a shot.

Is it possible for you to point out the places where I would need to make changes for current CLOCs to use 2D detector weights?

pangsu0613, 2021-07-20:
https://github.com/pangsu0613/CLOCs/blob/e0987f33e0f87b729b5128e7a3c9cf94977a7d23/second/pytorch/models/voxelnet.py#L389
This is the place where 2D detection files are loaded.

closed 2021-07-21 by pangsu0613

abhishek-t-naive, 2021-07-27:
> Hello @abhishek-t-naive, thank you for your interests in CLOCs, the cascade-RCNN that I am using is from this repo: https://github.com/zhaoweicai/mscnn, it is a pretty old one, because when I [...]

We have closed the issue but I still had one doubt. You mentioned that you used MSCNN as a 2D detector. The paper mentions that for CLOCs SecCas you used Cascade RCNN but this repo https://github.com/zhaoweicai/mscnn contains the implementation of MSCNN. Am I missing anything here?

pangsu0613, 2021-07-27:
The repo contains both Cascade RCNN and MSCNN. check here: https://github.com/zhaoweicai/mscnn/tree/master/examples/kitti_car

abhishek-t-naive, 2021-07-27:
Great. I just found it. 
I have one more query on the same topic:

I am trying to train Cascade RCNN on KITTI dataset. And for that, I am using the implementation provided by [mmdetection](https://github.com/open-mmlab/mmdetection). Since their model is trained on COCO, I will have to fine-tune the same for KITTI.  There are two things unique to the CLOCs requirement of box generated by 2d Detection:

1. Sigmoid scores: If I am not wrong I will have to modify the model provided to generate sigmoid scores as I suppose it doesn't do it at this point (it provide scores between -10 and 10)
2. And, since CLOCs only deals with the detection of cars, do I need to change the training process somehow to only deal with cars as I see that all the 2d detection files in CLOCs have CAR as the class

pangsu0613, 2021-07-27:
1. The simple solution answer is yes, just adding a sigmoid function after the scores (-10 to 10) is good to go. The code that I provided uses sigmoid score, but you don't have to, I have tested other type of scores as well, if you want to test other scores, you need to make some modifications in the code. 
2. The current CLOCs codebase is for single class, and uses CAR as an example. For your 2D detections, you can have other classes, but CLOCs will only read CAR class.

abhishek-t-naive, 2021-07-27:
That sounds good. 
Also, do you plan to open source the trained weights of the camera branch for the CLOC_Sec_cas network?

pangsu0613, 2021-07-27:
I used the pre-trained weight that they provided. You can download the weights from their repo.