pangsu0613/CLOCs #42: question about inference speed

RickOnEarth, 2021-08-05:


RickOnEarth, 2021-08-05:
@pangsu0613 Hi, I have tested the code on a Tesla V100 GPU. When I evaluate the kitti dataset, the inference speed is only about 2 fps. Is it normal? 

    for example in iter(eval_dataloader):
        if measure_time:
            prep_times.append(time.time() - t2)
            print("each prep time: ", (time.time() - t2)*1000)

a) I add the last line to see the data prep time. Usually it takes more than 15ms, while in SECOND code it only takes 3~5ms. 
The time should be the same, right? Since in this step, the data loaded are same in both CLOCs and SECOND. Could you explain the possible reason?
b) I also tested the function _predict_kitti_to_file(...) in train.py/evaluate(). Usually this function takes around 500 ms, in which net(example,detection_2d_path) is more than 300 ms. In the paper, you said CLOCs will not extend the lantency very much, but why it takes so long to do the inference. Maybe I did something wrong when I ran the code, could you please  help me to find the reason? How can I run the code as fast as you showed in the paper?

pangsu0613, 2021-08-06:
Hello @RickOnEarth , we have tested on 1080Ti, and it can run faster than 10Hz, for your situation, 2Hz is abnormal. Have you tested the running speed for SECOND only?

RickOnEarth, 2021-08-09:
@pangsu0613 Yes, I have tested SECOND in the same enviroment, and the running speed is about 25Hz. It is really confusing, I will continue to find the reason. Besides, you also used pointpillars as the basline for CLOCs, right? Could you please share the code? I am going to do some real-time applications with Pointpillrs based fusion network,  it will be appreciated if i can get the help from you.

pangsu0613, 2021-08-10:
There are some operations of moving data between GPU and CPU, maybe that's the issue causing the latency. For testing PointPillars, you could replace the SECOND network with PointPillar network, PointPillar also uses SECOND codebase. And for some other tips on increasing the running speed, you could refer to https://github.com/pangsu0613/CLOCs/issues/33

RickOnEarth, 2021-11-09:
@pangsu0613 Hi Dr. Pang, I have found the place where the code drags the speed down. In box_torch_ops.py/lidar_to_camera(), the original code 
`points = torch.cat([points, torch.ones(num_points, 1).type_as(points)], dim=-1)`
is very slow. The function torch.ones().type_as() takes more than 100ms . After I changed it to
 `points = torch.cat([points, torch.ones(num_points, 1, dtype=points.dtype, device=points.device)], dim=-1)` ,   
it costs only 0.2ms. This is weird, maybe it is relative to my environment or the version of the pytorch that I use. This problem also happens in the function project_to_image().   
Besides, in voxelnet.py, tensor_index1 is declared as:
`tensor_index1 = np.zeros((900000,2),dtype=box_2d_preds.detach().cpu().numpy().dtype)`, whose type is float32, 
 later the type is transformed to LongTensor with
`tensor_index_tensor = torch.LongTensor(tensor_index)`.
This declaration and transformation from 32 bits to 64 bits costs more than 100ms. After I changed it to
`overlaps1 = np.zeros((900000,4),dtype=np.float32)
tensor_index1 = np.zeros((900000,2),dtype=np.int64)
...
tensor_index_tensor = torch.LongTensor(tensor_index)`,  
the speed increased. Also this problem may only relate to  my environment. If anybody faces the same problem, I hope this comment could be a reference.
  
Other questions:
I have tested the inference speed after I modified the code.  With Tesla v100 and 4 core 2.1Ghz cpu, the inference speed of SECOND based CLOCs is about 56ms，the processing time  with function train_stage_2() is about18ms. I also replaced the SECOND network with Pointpillars network, the function train_stage_2() takes about 21ms, with total inference time about 41ms. Do you think the processing time of train_stage_2() is resonable? How is it in your case? I want to do realtime application with pointpillars based CLOCs, so I want to cut the processing time as much as possible.

pangsu0613, 2021-11-09:
Hello @RickOnEarth , thank you very much for your valuable feedback! There are multiple places where CLOCs is moving the variables between GPU and CPU back and forth which may drag down the speed, I have marked down the places you mentioned. 
One way to speed the things up significantly is to set a threshold for the 3D detector, for example 0.05 or 0.1, in SECOND, originally there are 70400 detection candidates (I believe PointPillars has similar numbers), but most of them have very very low scores and are false positives, I remembered setting a 0.05 or 0.1 threshold can reduce this number into hundreds or thousands (sorry I foget the exact number), this can significantly increase the speed but also remains a competitive performance. Then you can also change the '900000' shown above into much smaller numbers, this '900000' is the upper bound for the potential number of 2D-3D pairs.

RickOnEarth, 2021-11-10:
@pangsu0613 May I ask, how much time does the function train_stage_2() cost in your case? I would like to make a comparison.