pangsu0613/CLOCs #45: Training just the SECOND detector

abhishek-t-naive, 2021-08-10:
Hi
I have made changes to the SECOND part of CLOCS and I would like to train the network with those changes. Is it possible for me to do that using

`python ./pytorch/train.py train --config_path=./configs/car.fhd.config --model_dir=/dir/to/your_model_dir`
If it is can you let me know what changes I would have to do?

Thank you

pangsu0613, 2021-08-10:
Hello @abhishek-t-naive,
For training SECOND and then using it for CLOCs, I suggest using the original SECOND-V1.5 codebase instead of CLOCs codebase, you can follow the instructions here: https://github.com/traveller59/second.pytorch/tree/v1.5. Since you have already configured your environment for CLOCs, it would be very easy for you to run SECOND-V1.5 (remember to modify your PYTHONPATH).
The reason is that although CLOCs codebase is based on SECOND, but there are significant changes compared to original SECOND, and it is made to train CLOCs fusion network, not SECOND network. It is not trivial to make modifications on CLOCs to make it train SECOND network.

abhishek-t-naive, 2021-08-10:
Thank you for your quick response. 
My intention is to use the exact SECOND configuration present in CLOCS which includes using things like `VoxelFeatureExtractorV3` for voxel feature extraction, `SpMiddleVision` as middle feature extractor and `RPNV2` as RPN. But these are not present in the original SECOND codebase. 

1. Do you recommend that I do these changes (and others) to basically make the original SECOND codebase similar to the SECOND code used in CLOCS?

2. I am trying to retrain the network because I am trying to re-write the blocks like 
`block = Sequential(
                nn.ZeroPad2d(1),
                Conv2d(
                    in_filters[i], num_filters[i], 3, stride=layer_strides[i]),
                BatchNorm2d(num_filters[i]),
                nn.ReLU(),
            )`

into

`self.zero = nn.ZeroPad2d(1)
self.conv1 = conv2d(in_filters[i], num_filters[i], 3, stride=layer_strides[i])
self.batchnorm = BatchNorm2d(num_filters[i])
self.relu = nn.RELU()`

So these changes mean that I can't use the already provided weights. If you have a better suggestion of doing these changes and not requiring training please let me know.

pangsu0613, 2021-08-14:
Hello @abhishek-t-naive , sorry for the late response. 
The SECOND-V1.5 configuration present in CLOCs is the original SECOND-V1.5 configuration in SECOND-V1.5 codebase, I didn't modify SECOND-V1.5 in CLOCs. You could have a look at the original SECOND-V1.5 codebase (https://github.com/traveller59/second.pytorch/tree/v1.5). The VoxelFeatureExtractorV3 and SpMiddleVision you mentioned is implemented and applied in SECOND-V1.5, you could check their config file here: https://github.com/traveller59/second.pytorch/blob/v1.5/second/configs/car.fhd.config
I am not fully understand your intention, please let me know if your have further questions.

closed 2021-09-09 by pangsu0613