pangsu0613/CLOCs #52: error：bboxAP =0,something wrong but I don't no where .I just follow the readme.md introduction. Need help please.

Baser-Bai, 2021-11-25:
state.func_ir.loc))
Car AP@0.70, 0.70, 0.70:
bbox AP:0.00, 0.00, 0.00
bev  AP:0.00, 0.00, 0.00
3d   AP:0.00, 0.00, 0.00
Car AP@0.70, 0.50, 0.50:
bbox AP:0.00, 0.00, 0.00
bev  AP:0.00, 0.00, 0.00
3d   AP:0.00, 0.00, 0.00

pangsu0613, 2021-11-25:
Hello @Baser-Bai , there are multiple reasons that could cause this, could you show me your terminal output?

Baser-Bai, 2021-11-25:
> Hello @Baser-Bai , there are multiple reasons that could cause this, could you show me your terminal output?
Thank you for so quickly reply and the output as follow:
'''

(BR_CLOCs) vci-1@vci1:~/BR/FUSION_CAM_LADAR/CLOCs-master$ python train.py train --config_path=second/configs/car.fhd.config --model_dir=mymodel
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numba/cuda/decorators.py:110: NumbaDeprecationWarning: Eager compilation of device functions is deprecated (this occurs when a signature is provided)
  warn(NumbaDeprecationWarning(msg))
2d detection path: /home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/d2_detection_data/cascade_rcnn_sigmoid_data/data
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from mymodel/adam_optimizer-3712.tckpt
{'Car': 5}
[-1]
load 14357 Car database infos
load 2207 Pedestrian database infos
load 734 Cyclist database infos
load 1297 Van database infos
load 56 Person_sitting database infos
load 488 Truck database infos
load 224 Tram database infos
load 337 Misc database infos
After filter database:
load 10520 Car database infos
load 2104 Pedestrian database infos
load 594 Cyclist database infos
load 826 Van database infos
load 53 Person_sitting database infos
load 321 Truck database infos
load 199 Tram database infos
load 259 Misc database infos
remain number of infos: 3712
remain number of infos: 3769
WORKER 0 seed: 1637764427
WORKER 1 seed: 1637764428
WORKER 2 seed: 1637764429
/home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/second/core/geometry.py:146: NumbaWarning: 
Compilation is falling back to object mode WITH looplifting enabled because Function "points_in_convex_polygon_jit" failed type inference due to: No implementation of function Function(<built-in function getitem>) found for signature:
 
 >>> getitem(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))
 
There are 22 candidate implementations:
   - Of which 20 did not match due to:
   Overload of function 'getitem': File: <numerous>: Line N/A.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    No match.
   - Of which 2 did not match due to:
   Overload in function 'GetItemBuffer.generic': File: numba/core/typing/arraydecl.py: Line 162.
[... 474 lines cut ...]
bev  AP:0.00, 0.00, 0.00
3d   AP:0.00, 0.00, 0.00

Traceback (most recent call last):
  File "train.py", line 928, in <module>
    fire.Fire()
  File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/fire/core.py", line 141, in Fire
    component_trace = _Fire(component, args, parsed_flag_args, context, name)
[... cut ...]
    gt_annos, dt_annos, current_classes, overlap_ranges, compute_aos)
  File "/home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/second/utils/eval.py", line 828, in do_coco_style_eval
    min_overlaps[:, i, j] = np.linspace(*overlap_ranges[:, i, j])
  File "<__array_function__ internals>", line 6, in linspace
  File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numpy/core/function_base.py", line 120, in linspace
    num = operator.index(num)
TypeError: 'numpy.float64' object cannot be interpreted as an integer
'''

pangsu0613, 2021-11-25:
It is hard to see the problem. Could you just run evaluation with the pre-trained weights?

pangsu0613, 2021-11-25:
Also, you can add the things that I mentioned in the readme to hide these numbawarnings. and refer https://github.com/pangsu0613/CLOCs/issues/38#issuecomment-873198717 to fix the final TypeError.

Baser-Bai, 2021-11-25:
> It is hard to see the problem. Could you just run evaluation with the pre-trained weights
sorry.so long time to reply with you.I do only the eval and get the result as follow but with a problem named 
'''numpy.float64' "
'''
_CLOCs) vci-1@vci1:~/BR/FUSION_CAM_LADAR/CLOCs-master$ python train.py evaluate --config_path=./second/configs/car.fhd.config --model_dir=/home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/model/CLOCs_SecCas_pretrained  --measure_time=True --batch_size=1
Predict_test:  False
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
load existing model for fusion layer
Restoring parameters from /home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/model/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
[100.0%][===================>][10.09it/s][06:11>00:00]     
generate label finished(10.13/s). start eval:
validation_loss: 349.5329393248686
avg example to torch time: 3.058 ms
avg prep time: 7.142 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:29.31, 26.88, 23.52
bev  AP:1.88, 1.78, 1.45
3d   AP:0.48, 0.36, 0.27
aos  AP:16.74, 14.90, 13.03
Car AP@0.70, 0.50, 0.50:
bbox AP:29.31, 26.88, 23.52
bev  AP:8.64, 8.17, 6.76
3d   AP:5.69, 5.24, 4.22
aos  AP:16.74, 14.90, 13.03

Traceback (most recent call last):
  File "train.py", line 929, in <module>
    fire.Fire()
  File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/fire/core.py", line 141, in Fire
    component_trace = _Fire(component, args, parsed_flag_args, context, name)
  File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/fire/core.py", line 471, in _Fire
    target=component.__name__)
  File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/fire/core.py", line 681, in _CallAndUpdateTrace
    component = fn(*varargs, **kwargs)
  File "train.py", line 704, in evaluate
    result = get_coco_eval_result(gt_annos, dt_annos, class_names)
  File "/home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/second/utils/eval.py", line 1017, in get_coco_eval_result
    gt_annos, dt_annos, current_classes, overlap_ranges, compute_aos)
  File "/home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/second/utils/eval.py", line 828, in do_coco_style_eval
    min_overlaps[:, i, j] = np.linspace(*overlap_ranges[:, i, j])
  File "<__array_function__ internals>", line 6, in linspace
  File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numpy/core/function_base.py", line 120, in linspace
    num = operator.index(num)
TypeError: 'numpy.float64' object cannot be interpreted as an integer
'''

Baser-Bai, 2021-11-25:
> > It is hard to see the problem. Could you just run evaluation with the pre-trained weights
> > sorry.so long time to reply with you.I do only the eval and get the result as follow but with a problem named
> > '''numpy.float64' "
> > '''
> > _CLOCs) vci-1@vci1:~/BR/FUSION_CAM_LADAR/CLOCs-master$ python train.py evaluate --config_path=./second/configs/car.fhd.config --model_dir=/home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/model/CLOCs_SecCas_pretrained  --measure_time=True --batch_size=1
> > Predict_test:  False
> > sparse_shape: [  41 1600 1408]
> > num_class is : 1
> > load existing model
> > load existing model for fusion layer
> > Restoring parameters from /home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/model/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
> > remain number of infos: 3769
> > Generate output labels...
> > [100.0%][===================>][10.09it/s][06:11>00:00]
> > generate label finished(10.13/s). start eval:
> > validation_loss: 349.5329393248686
> > avg example to torch time: 3.058 ms
> > avg prep time: 7.142 ms
> > Car AP@0.70, 0.70, 0.70:
> > bbox AP:29.31, 26.88, 23.52
> > bev  AP:1.88, 1.78, 1.45
> > 3d   AP:0.48, 0.36, 0.27
> > aos  AP:16.74, 14.90, 13.03
> > Car AP@0.70, 0.50, 0.50:
> > bbox AP:29.31, 26.88, 23.52
> > bev  AP:8.64, 8.17, 6.76
> > 3d   AP:5.69, 5.24, 4.22
> > aos  AP:16.74, 14.90, 13.03
> 
> Traceback (most recent call last): File "train.py", line 929, in fire.Fire() File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/fire/core.py", line 141, in Fire component_trace = _Fire(component, args, parsed_flag_args, context, name) File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/fire/core.py", line 471, in _Fire target=component.**name**) File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/fire/core.py", line 681, in _CallAndUpdateTrace component = fn(*varargs, **kwargs) File "train.py", line 704, in evaluate result = get_coco_eval_result(gt_annos, dt_annos, class_names) File "/home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/second/utils/eval.py", line 1017, in get_coco_eval_result gt_annos, dt_annos, current_classes, overlap_ranges, compute_aos) File "/home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/second/utils/eval.py", line 828, in do_coco_style_eval min_overlaps[:, i, j] = np.linspace(*overlap_ranges[:, i, j]) File "<**array_function** internals>", line 6, in linspace File "/home/vci-1/anaconda3/envs/BR_CLOCs/lib/python3.7/site-packages/numpy/core/function_base.py", line 120, in linspace num = operator.index(num) TypeError: 'numpy.float64' object cannot be interpreted as an integer '''

this problem is caused by numpy version. I got it .

closed 2021-11-25 by Baser-Bai

pangsu0613, 2021-11-25:
Hi @Baser-Bai ,Based on your newest terminal output, it seems that you didn't setup the original pre-trained SECOND model properly. There should be a line saying "Restoring parameters from ../model_dir/voxelnet-30950.tckpt" below "loading existing model" and before "load existing model for fusion layer".

Baser-Bai, 2021-11-25:
> Restoring parameters from

BR_CLOC) vci-1@vci1:~/BR/FUSION_CAM_LADAR/CLOCs-master$ python train.py evaluate --config_path=./second/configs/car.fhd.config --model_dir=./model/CLOCs_SecCas_pretrained  --measure_time=True --batch_size=1
Predict_test:  False
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
load existing model for fusion layer
Restoring parameters from model/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
[100.0%][===================>][9.87it/s][06:20>00:00]      
generate label finished(9.90/s). start eval:
validation_loss: 377.48785447251987
avg example to torch time: 2.935 ms
avg prep time: 7.661 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:34.25, 31.04, 27.11
bev  AP:1.30, 1.32, 1.10
3d   AP:0.21, 0.19, 0.17
aos  AP:20.31, 17.47, 15.35
Car AP@0.70, 0.50, 0.50:
bbox AP:34.25, 31.04, 27.11
bev  AP:6.84, 7.39, 6.25
3d   AP:4.30, 4.38, 3.67
aos  AP:20.31, 17.47, 15.35

Car coco AP@0.50:0.05:0.95:
bbox AP:25.07, 23.07, 20.49
bev  AP:1.95, 2.07, 1.76
3d   AP:0.94, 0.95, 0.79
aos  AP:14.81, 13.00, 11.59

Thanks again. With your help I got the hoping   result . @pangsu0613 
'''
(BR_CLOC) vci-1@vci1:~/BR/FUSION_CAM_LADAR/CLOCs-master$ python train.py evaluate --config_path=./second/configs/car.fhd.config --model_dir=/home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/model/CLOCs_SecCas_pretrained  --measure_time=True --batch_size=1
Predict_test:  False
sparse_shape: [  41 1600 1408]
num_class is : 1
load existing model
Restoring parameters from model/second_model/voxelnet-30950.tckpt
load existing model for fusion layer
Restoring parameters from /home/vci-1/BR/FUSION_CAM_LADAR/CLOCs-master/model/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
remain number of infos: 3769
Generate output labels...
[100.0%][===================>][11.39it/s][05:27>00:00]     
generate label finished(11.50/s). start eval:
validation_loss: 0.14486018488443173
avg example to torch time: 2.864 ms
avg prep time: 7.954 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:99.32, 93.75, 92.88
bev  AP:96.54, 92.35, 89.41
3d   AP:92.76, 82.82, 77.74
aos  AP:99.16, 93.29, 92.10
Car AP@0.70, 0.50, 0.50:
bbox AP:99.32, 93.75, 92.88
bev  AP:99.38, 96.27, 95.56
3d   AP:99.55, 96.18, 95.34
aos  AP:99.16, 93.29, 92.10
'''