pangsu0613/CLOCs #53: Validation AP result is lower to expected

Typolo, 2021-11-25:
Hi, thank you for your work; 
I follow the instructions in README.md to run the evaluation code,  but result is lower to expected.
The evaluation result:
      Predict_test:  False
      sparse_shape: [  41 1600 1408]
      num_class is : 1
      load existing model
      Restoring parameters from ../model_dir/voxelnet-30950.tckpt
      load existing model for fusion layer
      Restoring parameters from /wyy/CLOCs-master/CLOCs_SecCas_pretrained/fusion_layer-11136.tckpt
      remain number of infos: 3769
      Generate output labels...
      [100.0%][===================>][8.93it/s][07:02>00:00]
      generate label finished(8.90/s). start eval:
      validation_loss: 0.18061617695790105
      avg example to torch time: 2.237 ms
      avg prep time: 23.910 ms

      Car AP@0.70, 0.70, 0.70:
      bbox AP:96.47, 93.23, 90.20
      bev  AP:92.84, 88.53, 85.40
      3d   AP:82.89, 75.80, 71.95
      aos  AP:95.98, 91.94, 88.43

      Car AP@0.70, 0.50, 0.50:
      bbox AP:96.47, 93.23, 90.20
      bev  AP:98.35, 95.64, 92.70
      3d   AP:96.70, 95.23, 92.34
      aos  AP:95.98, 91.94, 88.43
I can't figure why and look forward to your reply,thank you.

pangsu0613, 2021-11-25:
Hello @Typolo , thank you for your interests in CLOCs, may I know if you modify the code, or setup the 2D detection path correctly in the config file?
If all above are correctly setup, you could comment this line: https://github.com/pangsu0613/CLOCs/blob/205529b78ca2a85c9d80b653d2467501d51baf20/second/pytorch/train.py#L466   
, this will disable CLOCs and run the original SECOND, then run evaluation and see if you can get good results for original SECOND. So we can see if it is other software version problem.

Typolo, 2021-11-26:
Hi @pangsu0613,thank you. I did't modify the code and the 2d detection path is set correctly as README.md;
I try comment the line in train.py, and the evaluation result is:

validation_loss: 0.18061617695790105
avg example to torch time: 1.908 ms
avg prep time: 39.500 ms
Car AP@0.70, 0.70, 0.70:
bbox AP:85.80, 85.49, 83.77
bev  AP:82.11, 82.12, 80.67
3d   AP:73.12, 69.24, 67.43
aos  AP:85.48, 84.78, 82.81

Car AP@0.70, 0.50, 0.50:
bbox AP:85.80, 85.49, 83.77
bev  AP:88.78, 88.97, 88.98
3d   AP:86.98, 88.53, 88.31
aos  AP:85.48, 84.78, 82.81

Also, the code is tested on python3.7, pytorch1.4.0 , Ubuntu 18.04; Is the version a problem?

pangsu0613, 2021-11-30:
This is odd, your environment looks fine. I have tested CLOCs using python 3.6/3.7, pytorch 1.1/1.7, ubuntu 16.04/18.04. It seems like there is something wrong with the original SECOND setup, did you install the specific spconv version that I recommended in the readme? Could you show me your config file?

Typolo, 2021-12-01:
I misunderstanding the part about velodyne_reduced. I thought it need to be empty. Now, I can get same right result, thank you.

closed 2021-12-01 by Typolo