pangsu0613/CLOCs #54: CUDA out of memory in training

Rorywh, 2021-11-30:
Hi, I tried to train the model in my PC (Nvidia 1080Ti). But, it raised the following issues:

RuntimeError: CUDA out of memory. Tried to allocate 70.00 MiB (GPU 0; 10.92 GiB total capacity; 9.29 GiB already allocated; 59.50 MiB free; 9.40 GiB reserved in total by PyTorch) If reserved memory is >> allocated memory try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF

Is there any way to train the model on my PC. My environments are as follows:
Nvidia 1080Ti, CUDA 11.3,  Pytorch 11.3, Python 3.9,  spconv 1.0 and second 1.5.

Rorywh, 2021-12-01:
Thanks for your advice @pangsu0613. I have checked the memory usages of my GPU using "nvidia-smi". When only basic processes is running, it still raised the "CUDA out of Memory".  Bellow are the processes :

>  0   N/A  N/A      1151      G   /usr/lib/xorg/Xorg                 18MiB |
|    0   N/A  N/A      1252      G   /usr/bin/gnome-shell               14MiB |
|    0   N/A  N/A      1445      G   /usr/lib/xorg/Xorg                 81MiB |
|    0   N/A  N/A      1591      G   /usr/bin/gnome-shell               71MiB

The raised issues are as follows:

> RuntimeError: CUDA out of memory. Tried to allocate 54.00 MiB (GPU 0; 10.92 GiB total capacity; 9.52 GiB already allocated; 60.62 MiB free; 9.63 GiB reserved in total by PyTorch) If reserved memory is >> allocated memory try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF

The memory for allocation is slightly different from the previous one. 
And I wonder if there is any other way to solve this problem. But the batch_size in "train_input_reader" of the "car.fhd.config" is already "1".

pangsu0613, 2021-12-01:
Running this code on my 1080Ti only uses around 4 to 5 GB GPU memory. What version of pytorch are you using?

Rorywh, 2021-12-02:
Thanks a lot! It works! I used 1.7.1 pytorch instead of the 1.9.0 and it no longer raise the issues.

closed 2021-12-02 by Rorywh

KexuanXia, 2024-06-30:
Hi, 
My CUDA version is 11.3 just the same as yours, so I wonder if it's possible to install spconv 1.0 under CUDA 11.3, because I am facing some problems when installing spconv 1.0.

Thank you in advance.

KexuanXia, 2024-07-02:
@Rorywh 
Hi,
my CUDA version is also 11.3 and met the same problem(CUDA out of memory) as you. I see that you used pytorch 1.7.1 version to make it work.
But I have checked pytorch official release, pytorch 1.7.1 doesn't support CUDA11.3. How did you make it?

Thank you in advance.