pangsu0613/CLOCs #55: some qusetion about the dataset and the 2D detection candidates

hhl12138, 2022-01-09:
thank u so much，I think your article is very well written。
1.
  I changed the number of channels for the dataset from 4 to other number, do I need to re-operate on the dataset to get a .pkl file? If so, can I ask for advice on how to manipulate the dataset, or if you only need to modify part of the code?（I want to use the score from the 2D image segmentation as a feature of the lidar data）
2.
if i want to use other 2D detection candidates，how do i get it ? Please forgive me for asking such a question because I don't know much about it。
I hope you can answer my questions，thanks again。

pangsu0613, 2022-01-15:
Hello @hhl12138 , Thank you for your interests in CLOCs!

pangsu0613, 2022-01-15:
Hello @hhl12138 , Thank you for your interests in CLOCs!
1. You don't need to get new pkl file, just modify part of the code is sufficient. You need to change channel number 4 into your custom channel number. My apologize for hardcoding the channel number in the code. Here are some places I remember that need changes: /CLOCs/blob/master/second/pytorch/models/voxelnet.py, around line 521, for variable iou_test_tensor, non_empty_iou_test_tensor. /CLOCs/blob/master/second/utils/eval.py, line 127, function build_stage2_training, this is the function used to generate input tensor, so channel number is considered here as well.
2. For most of the neural network-based detectors, they have a post processing stage where NMS and score thresholdinng are done. You could disable the NMS function or set the score threshold into a very small value or zero to get the detection candidates. But as a start, I would suggest just modify the score threshold since it would be easier and won't affect the final performance much. Or, if you just want to run some quick tests, not modifying anything for the 2D detector is also fine, just take the output 2D detections (final 2D bounding boxes and confidence scores) and fuse it with 3D detections, CLOCs can also works, but not as good as using candidates, but the result is still acceptable (better than original 3D detector). I have tested it before. 

Let me know if you have more questions.

hhl12138, 2022-01-15:
@pangsu0613  thanks for your reply
 My current situation is like this--I increased the number of channels by reconstructing training dataset of velodyne from 4 to 6，That is to say，I regenerated the  .bin file，Do I not have to modify the num_input_features and num_point_features in the config file? for the : /CLOCs/blob/master/second/pytorch/models/voxelnet.py,I change the 4 to 6, am i right？ for the /CLOCs/blob/master/second/utils/eval.py, line 127,I have no idea about how to change it . 
another question , why dont u use BatchNormal in the fusion layer?
looking forward to your reply，thanks again

pangsu0613, 2022-01-16:
@hhl12138 , Ah, I see your point. That's a different story, for your case, I think what you want to do is training a new 3D detector that takes 6 channel point cloud, you probably need to re-train SECOND (with modifications of course, because original SECOND takes 3 or 4 channels point cloud) or other SOTA LiDAR-based networks (such as PV-RCNN and so on). CLOCs is a detection level fusion framework, it fuses 3D detections (3D bounding boxes) and 2D detections (2D bounding boxes). So, after you train your new 3D detector with 6 channel point cloud, CLOCs can help you fuse your output 3D detections and 2D detections, but you need to train your new 3D detectors first.
Batch normalization is usually used for training very deep neural networks that standardizes the inputs to a layer for each mini-batch. However, CLOCs fusion is a shallow network, so there is no need to use BatchNorm. I have tested BatchNorm in CLOCs, it doesn't affect the final performance.

hhl12138, 2022-01-16:
THANKS AGAIN!!

closed 2022-01-16 by hhl12138