pangsu0613/CLOCs #56: About merging 2D with pointpillars using Clocs

hhl12138, 2022-01-21:
HELLO,sir，take the liberty to disturb you，I have read your article on Clocs，now ,i want to merge 2D with pointpillars using Clocs,what do i need to do?Let me first tell you what I think, can you see if it works? First, I use pointpillars to train to get ckpt, change the config file to pointpillars, put pointpillars.py into second/pytorch/models, do I need to change anything else? Maybe it's an abrupt question，I would be very grateful if you could answer, I didn't use github for questions because of network issues。Besides，I would really appreciate it if you could send me a repo of pointpillars that has been integrated。
Look forward to your reply

Joiequ, 2022-03-12:
Hello, sir or miss, have you solved this problem?

hhl12138, 2022-03-12:
> Hello, sir or miss, have you solved this problem?

no，i dont have a idea