pangsu0613/CLOCs #57: questions about ONNX export

CrapbagMo, 2022-03-02:
hello @pangsu0613 ! When I tried to export this model to onnx (opset=9), I got this problem
`RuntimeError: Unsupported: ONNX export of index_put in opset 9`
I was wondering if the following code in fusion.py can be replaced by torch.scatter or other op supported by onnx (opset=9)
\out_1[:, tensor_index[:, 0], tensor_index[:, 1]] = x[0, :, 0, :]\

pangsu0613, 2022-03-25:
Hello @CrapbagMo , sorry for the late response. I apologies that I don't have too much experience with model transformation. The line of code you highlighted is puting the x into out_1 based on the indices, which is tensor_index.

zizo1111, 2022-06-21:
Hi @CrapbagMo , have you reached any thing with the exportation to onnx. I am currently trying to do the same thing but i am stuck.