pangsu0613/CLOCs #59: Can you provide more fusion models at your paper

Joeless, 2022-03-05:
Hi~ @pangsu0613 
I like your works~
I read your paper and found that you have done a lot of model experiments, but your repo only provides Second+Cascade-RCNN
Can you provide more fusion models at your paper？

Joeless, 2022-03-07:
Hi~
Can you provide 2D det model?   
I think the 2d det model should  add  in this repo

pangsu0613, 2022-03-14:
Thank you for your interests in CLOCs! Replacing 2D detections is quite easy for the current codebase, you could just replace the 2D detection filtes with yours, but the files should be in KITTI format.

Joeless, 2022-03-15:
> Thank you for your interests in CLOCs! Replacing 2D detections is quite easy for the current codebase, you could just replace the 2D detection filtes with yours, but the files should be in KITTI format.

Thank you for your answer~
2D detection result is easy to replace
I find your recently work about fast-clocs，congratulations on your interesting article , When will this code be released？

pangsu0613, 2022-03-25:
I have started working on this, I plan to release it in April. thank you for your interests.

closed 2022-03-25 by pangsu0613