pangsu0613/CLOCs #60: TypeError: unsupported array index type list(int64)<iv=None> in Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>)

SunshineKK456, 2022-03-09:
hello,I meet some troubles when I trained the CLOCs:

There are 22 candidate implementations:
   - Of which 20 did not match due to:
   Overload of function 'getitem': File: <numerous>: Line N/A.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    No match.
   - Of which 2 did not match due to:
   Overload in function 'GetItemBuffer.generic': File: numba/core/typing/arraydecl.py: Line 162.
     With argument(s): '(array(float32, 3d, C), Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>))':
    Rejected as the implementation raised a specific error:
      TypeError: unsupported array index type list(int64)<iv=None> in Tuple(slice<a:b>, list(int64)<iv=None>, slice<a:b>)

When I finished 3700 steps,I meet the above question,Can you give me some help or advice？Thanks.

SunshineKK456, 2022-03-09:
I find some solutions on the Internet,just like change the function predict_v2:

opp_labels=(box_preds[...,-1] >0) ^dir_lables.byte()  →  opp_labels=(box_preds[...,-1] >0) ^dir_lables.to(torch.bool)

But it doesn't work.....

pangsu0613, 2022-03-25:
Hello @SunshineKK456 , I think I need more information regarding your error. May I know in what situation you encountered these errors? are you training with other 2D or 3D detections? Could you paste all the error message from the terminal? Thank you.

SunshineKK456, 2022-03-25:
@pangsu0613  hello, I have solved the trouble. Thanks for your excellent work !

SunshineKK456, 2022-03-26:
@pangsu0613 I see a same issue before, and followed the steps. Someone suggest me to change the labels‘ data type. As you can see, I tried to change the 'byte()' into the 'bool' in pred(there are no mistakes in the training), but it doesn't work. By the way, I like saber too.

closed 2022-03-26 by SunshineKK456

IsFaqq, 2023-04-10:
@SunshineKK456 hello,I had the same problem, how did you end up solving it? I also tried to change the 'byte()' into the 'bool' in pred,but it doesn't work.