pangsu0613/CLOCs #63: Pedestrian val: Bbox, BEV,3D all are 0.0

Rorywh, 2022-03-23:
Hello, I tried to use the pretrained "mscnn_ped_cyc_trainval_sigmoid_data_scale_1000" and "second_pedestrian_model" to train the model. I have followed the instructions to change some codes in the pedestrian.fhd.config, train.py and the voxelnet.py. The training process is normal except the eval results:

now it is 74150 steps  and the cls_loss is : tensor(0.0982, device='cuda:0', grad_fn=<DivBackward0>) learning_rate:  6.089296127003236e-08
now it is 74200 steps  and the cls_loss is : tensor(0.0950, device='cuda:0', grad_fn=<DivBackward0>) learning_rate:  3.6271128087118313e-08
#################################
# EVAL
#################################
Generate output labels...
validation_loss: 0.008988450088923213
generate label finished(16.10/s). start eval:
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:0.00, 0.00, 0.00
bev  AP:0.00, 0.00, 0.00
3d   AP:0.00, 0.00, 0.00
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:0.00, 0.00, 0.00
bev  AP:0.00, 0.00, 0.00
3d   AP:0.00, 0.00, 0.00

So, can you help me with this? Thanks for your attention.

Rorywh, 2022-03-25:
> 

I have found the reason, and it can output the val results. But the pedestrian 3D AP is lower than expected.

pangsu0613, 2022-03-25:
Hello @Rorywh, thank you for your interests in CLOCs, if the evaluation results are all 0.0, it could be the models are not correctly loaded, could you show more information from the terminal output?

Rorywh, 2022-03-29:
> Hello @Rorywh, thank you for your interests in CLOCs, if the evaluation results are all 0.0, it could be the models are not correctly loaded, could you show more information from the terminal output?

Thanks for your help. Now the model can output the val results. But it seems that the val results is lower than expected when training and testing on pedestrian.

closed 2022-03-29 by Rorywh