pangsu0613/CLOCs #65: Is the SECOND code in your project the same as that of the original author?

xuV2, 2022-04-21:
If so, can I directly look at the code you wrote without knowing the SECOND code?

xuV2, 2022-04-21:
@pangsu0613

pangsu0613, 2022-04-21:
Hello @xuV2, thank you for your interests in CLOCs. It is a little bit hard to say, I did make modifications in the original SECOND code, but not the core model part. But if you have any questions regarding the code, please feel free to ask.

closed 2022-04-21 by pangsu0613

xuV2, 2022-04-22:
> Hello @xuV2, thank you for your interests in CLOCs. It is a little bit hard to say, I did make modifications in the original SECOND code, but not the core model part. But if you have any questions regarding the code, please feel free to ask.

Thank you for your reply！

xuV2, 2022-04-29:
Hi @pangsu0613 
I want to ask you two questions, 1.What is self.corner_points_feature  in fusion.py ？
2.Is the tensor_index with flag=0 in fusion.py  finally discarded ？Thank you.

pangsu0613, 2022-04-29:
1. the corner_points_feature is not used. I should have deleted it, sorry for the confusion. I used it to test with fusing with boounding boxes corner points coordinates, but the results turned to be not good, so I ended up with not using it.
2. tensor_index with flag = 0  means there are no overlaps between the 3D detections and 2D detections in the current frame, so we cannot learn anything, so they are not contribute in the training, or say discarded.

xuV2, 2022-05-06:
Excuse me again @pangsu0613 ，I would appreciate it if you could explain the code 
`all_3d_output_camera_dict, all_3d_output, top_predictions, fusion_input,tensor_index = net(example_torch,detection_2d_path)`  in train.py.