pangsu0613/CLOCs #67: How do you select the best CLOCs model when submitting to test server?

rkotimi, 2022-05-24:
Hi @pangsu0613 !I am still doing some research on CLOCs and Fast-CLOCs. Thank you for your nice works!
When submitting test set results to KITTI server, many works use all 7481 training samples to train the model. However, as mentioned in #46, the training process of CLOCs is unstable. I think it is hard to select a CLOCs model with good generalization performance without validation set. And I also find it hard to give a reasonable division of all the training samples by myself. For example, 80% for training and 20% for validation. I wonder how do you do it. Could you help me?

pangsu0613, 2022-05-30:
Hello @rkotimi, based on my experience, I suggest training CLOCs with trainval (3712 + 3769) dataset, but still validate on the 3769 validation set. I understand this is not a perfect solution, but the validation results will still give some sort of indication on the training. For my experiment, I trained it for 10 epochs (with trainval) and then submit.
I agree the training is not stable, but it is also not unstable. I did many experiments on KITTI with different 3D/2D detectors, and I found sometimes the performance varies, from my perspective, the main reason is that KITTI is still a relatively small dataset and the performance is quite sensitive to small changes in the network.

rkotimi, 2022-05-30:
Thank you for your suggestion. I agree that the small size of KITTI dataset is the main reason.

closed 2022-05-30 by rkotimi