pangsu0613/CLOCs #69: fuse other 2D detection

cxy-ccc, 2022-06-25:
I have fused other 2D detetions such as yolov5 and yolox, but AP is very low and close to 0. The 3D method is SECOND, and i used the  pretrain model mentioned in readme. I only replace the file "d2_detection-data" with yolov5 results in kitti label format. I have no idea about which step is wrong.

pangsu0613, 2022-06-25:
Hello @cxy-ccc, thank you for your interests in CLOCs. May I know which class are you trying to fuse, and what is the score scale  of your 2D detections, is it sigmoid score (between 0.0-1.0) or log score (between -inf to +inf)? Also, may I know if you can get the results that I provided in the repo?

cxy-ccc, 2022-06-28:
> Hello @cxy-ccc, thank you for your interests in CLOCs. May I know which class are you trying to fuse, and what is the score scale of your 2D detections, is it sigmoid score (between 0.0-1.0) or log [...]

I use the class "car" to fuse. And yolov5 is sigmoid score. I can restore your results with the methods mentioned in your paper (C-RCNN+SECOND in class car).

pangsu0613, 2022-06-28:
@cxy-ccc , everything looks right. If '2d detection path' is the only thing that you changed, could you manually check the  2d detection txt files? maybe compare it with the default ones (cascade-rcnn) and see if the number of colums, box encodings are the same? You could also share some samples here so I can also have a look.

cxy-ccc, 2022-06-28:
> @cxy-ccc , everything looks right. If '2d detection path' is the only thing that you changed, could you manually check the 2d detection txt files? maybe compare it with the default ones [...]

Thank you for your tips. The results are not familiar. I set the nms to 0. Maybe confidence should be adjusted?
___________________________________________________________________________________
car -1 -1 -10 27.1 171.61 226.1 297.61 -1 -1 -1 -1000 -1000 -1000 -10 0.9692
This is the content of 000001.txt. It only export one result. However, the results of c-rcnn is 
____________________________________________________________________________________
Car -1 -1 -10 387.97 181.29 425.20 202.74 -1 -1 -1 -1000 -1000 -1000 -10 0.8444 
Car -1 -1 -10 469.85 59.68 519.34 88.63 -1 -1 -1 -1000 -1000 -1000 -10 0.0275 
Car -1 -1 -10 599.46 163.49 629.96 190.04 -1 -1 -1 -1000 -1000 -1000 -10 0.0235 
Car -1 -1 -10 756.64 183.04 810.01 203.49 -1 -1 -1 -1000 -1000 -1000 -10 0.0064 
Car -1 -1 -10 41.09 129.28 118.02 180.63 -1 -1 -1 -1000 -1000 -1000 -10 0.0052 
Car -1 -1 -10 831.21 120.47 877.35 145.45 -1 -1 -1 -1000 -1000 -1000 -10 0.0018 
Car -1 -1 -10 1044.50 200.98 1110.56 234.57 -1 -1 -1 -1000 -1000 -1000 -10 0.0008 
Car -1 -1 -10 831.93 152.59 895.27 181.13 -1 -1 -1 -1000 -1000 -1000 -10 0.0006 
Car -1 -1 -10 378.20 59.09 449.14 89.29 -1 -1 -1 -1000 -1000 -1000 -10 0.0006 
Car -1 -1 -10 964.38 189.73 1041.31 223.06 -1 -1 -1 -1000 -1000 -1000 -10 0.0001 
Car -1 -1 -10 1062.60 198.48 1135.43 235.10 -1 -1 -1 -1000 -1000 -1000 -10 0.0001 
Car -1 -1 -10 810.65 148.52 904.82 189.11 -1 -1 -1 -1000 -1000 -1000 -10 0.0000

pangsu0613, 2022-06-28:
@cxy-ccc, I guess there are two potential issues. (1). The class label in the first colum should be 'Car', not 'car', you could  change  https://github.com/pangsu0613/CLOCs/blob/cad14fdc12392b9734d496e5d7782ae3ba200af5/second/pytorch/models/voxelnet.py#L393, replace 'Car' with 'car'. (2). I suggest visualize some of your yolo predictions on the image plane to see if these yolo boxes are correct. The box you showed for 000001.txt seems not right. Don't worry about the number of predictions in Cascade-RCNN, most of them have very low confidence score, it won't affect the results.

As for the nms setting for YOLO, usually there are 2 nms thresholds, one is nms IoU thresh, the other is nms score thresh. I guess you are talking about IoU threshold. The meaning for IoU thresh is that, any box overlap larger than the thresh will be discarded, so I think it should not be set to 0.0, that would result in output boxes with NO overlaps at all (any boxes with overlaps will be removed). If you want more predictions, the nms IoU thresh should be set to a large value, but based on my experience, I recommend 0.5 or 0.7, too large will also be harmful.

cxy-ccc, 2022-06-30:
> @cxy-ccc, I guess there are two potential issues. (1). The class label in the first colum should be 'Car', not 'car', you could change [...]

Thank you so much for your advice！！！The most critical problem should be the error of my labels. I checked the labels and AP can be displayed normally. For car,
__________________________________________________
bev  AP:89.39, 87.09, 84.96
3d    AP:89.36, 85.07, 84.76
And, actually the results I restored of your paper is very high and close to 99
bev  AP:99.32, 96.56, 93.74
3d   AP:99.51, 96.47, 93.62