pangsu0613/CLOCs #81: The result of your paper is different from the result of the code.

xuV2, 2022-10-12:
- Hi! @pangsu0613 ，I found that the results of your paper are not the same as the results of the code.Can you explain why?
-
[image]

[image]

dinghuixiang-github, 2022-11-09:
The screenshot of this paper is the KITTI test set result, and the code reasoning result is the KITTI validation set result. By the way, the validation set results of the code are generally similar to the validation set results in the paper.

xuV2, 2022-12-29:
> The screenshot of this paper is the KITTI test set result, and the code reasoning result is the KITTI validation set result. By the way, the validation set results of the code are generally similar to the validation set results in the paper.

Excuse me, how do I get the results of the test set? This problem has troubled me for a long time

dinghuixiang-github, 2022-12-29:
The inference on validation split can be implemented locally offline ,while the inference on test split can only be implemented on the kitti offical website, such as the "submit results" button below. I haven't submitted the test result, so I don't know more details.   
[image]

> Excuse me, how do I get the results of the test set? This problem has troubled me for a long time

xuV2, 2022-12-29:
Thank you very much for your reply, your answer is very helpful